stack-of-tasks/pinocchio #528: Downloading Pinocchio from release zone does not provide submodule

cmastalli, 2018-09-27:
Hi guys,

I noticed a little bug in the compilation of Pinocchio using release compress files. These compress files do not provide the information of the submodule. It's not a bit deal to put it manually however it creates confusion in the installation.

Best,
Carlos

nim65s, 2018-09-27:
Hi @cmastalli 

I don't understand your issue.
The content of the folders in the submodules seems to be inside the .tar.gz file.

Could you detail a bit more what do you need ?

cmastalli, 2018-09-27:
I didn't see them inside the zip files. I didn't try the other formats.

nim65s, 2018-09-27:
Now I understand :)

The last two archives, "Source code", in zip and tar.gz versions, are generated by Github for each tag, but doesn't work. 
This is why we release and upload the files like "pinocchio-1.3.1.tar.gz", which contains the content of the submodules, plus a few aditionnal stuff to make it an actual release, like a `.version` file.

Please use this one :)

closed 2018-09-27 by nim65s

cmastalli, 2018-09-27:
Thanks for clarifying this. Is there a way to disable the generation of this file for GitHub?

jcarpent, 2018-09-27:
No, there is no way to remove this automatic generation. Unfortunately.