stack-of-tasks/pinocchio #529: [bindings] Function naming inconsistencies between Python and C++

gabrielebndn, 2018-10-02:
The following methods have a Python name which is different from the C++ name:
- impulseDynamics (impactDynamics in C++)
- nle (nonLinearEffects in C++)

jcarpent, 2018-10-02:
I do not which one is the most correct between `{impulse,impact}Dynamics`. 
`nle` is just a shortcut for `nonLinearEffects`, so I think it can be kept and `nonLinearEffects` can be added.

cmastalli, 2018-10-02:
I vote for impulse

gabrielebndn, 2018-10-02:
Agreed for `nle`

gabrielebndn, 2018-10-02:
As for `{impulse,impact}Dynamics`, I had switched the two, it was actually `impact` in python and `impulse` in C++. I corrected now.

I also prefer `impulseDynamics`. Also, this way the C++ interface stays the same. Python's `impactDynamics` would become deprecated then

proyan, 2018-10-03:
For 'nle' and 'nonlineareffects', we have another nomenclature in robotwrapper, called 'bias'.
I would propose uniformizing this as well.

jcarpent, 2018-10-03:
I think `bias`should be made deprecated.

gabrielebndn, 2018-10-03:
I agree. What about `mass`?

commit 1eae3be8 (2018-10-30): Merge pull request #531 from gabrielebndn/devel

[bindings] Implement bindings, solve name inconsistencies, fix #529
commit 4fd3b190 (2018-10-30): [bindings] Implemented missing bindings, resolved naming inconsistencies, fix #529