stack-of-tasks/pinocchio #533: [python] python test folder is broken

gabrielebndn, 2018-10-03:
The whole python folder is a mess. Most tests seem to be broken.
- One test did not even parse properly because of a typo. That's the only thing I corrected, because it was too bad.
- Still, once the typo is corrected, tests.py does not even run to completion because some tests load the romeo model and the path is wrong, raising an uncaught exception. I corrected those errors, but I did not include the correction in this pull request, because I don't think romeo should be used in the tests anyway
- Once the above problems are corrected, lots of tests fail. I did not check the errors

By the way, I don't like the name "python", it's confusing, it is not clear that it only contains tests.

nim65s, 2018-10-03:
Clément Bonnefont already worked on that, but I can't find it again.
Making this work has been in my todolist for a long time, but I'm surprised that there is not already an issue about that.

jcarpent, 2018-10-04:
This is also due to the fact that Python is not more tested during the unit tests procedure.
It should be reactivated for producing much more stable releases.

jcarpent, 2018-10-17:
Solved by #550

commit 8e91554a (2018-10-17): Merge pull request #550 from gabrielebndn/topic/pythontest

[python] Fixed python tests and moved them into unit test, fix #533
commit dcc3b9cc (2018-10-24): Merge pull request #550 from gabrielebndn/topic/pythontest

[python] Fixed python tests and moved them into unit test, fix #533
closed 2018-10-17 by jcarpent