stack-of-tasks/pinocchio #534: Improving coverage

jcarpent, 2018-10-04:
The coverage of the project seems pretty good (91 %).
I think we can still improve by modifying the way the coverage is done.

I suggest to remove from coverage:
- the files related to the exposition of Pinocchio in Python (bindings/python/ folder)

I will then go back to the coverage status to look which part of the code is not covered by the unit tests.

@nim65s Can you remove the folder `bindings/python` from the coverage?

nim65s, 2018-10-04:
I can do that for now, but I think we should add those back when we will have the CI launch the python tests.

This way, we could also get the coverage of the .py files.

jcarpent, 2018-10-04:
Yes, it seems indeed a good solution.

jcarpent, 2018-10-05:
http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/coverage/ does not seem to reflect your modifications.

nim65s, 2018-10-05:
Because those modifications are not in the `master` branch of the `stack-of-tasks` repository of gitlab.
You can see the list of available organisation / project / branches here:
http://projects.laas.fr/gepetto/doc/

So, for now, they are available on
http://projects.laas.fr/gepetto/doc/gsaurel/pinocchio/devel/coverage/
And, because you merged those changes on github, and I mirrored that to gitlab, they will soon be available on
http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/devel/coverage/

jcarpent, 2018-10-05:
Thanks for this reply.
It appears then some files related to Python bindings remain active.
See for instance include/pinocchio/bindings/python/multibody/joint/joints-variant.hpp in http://projects.laas.fr/gepetto/doc/gsaurel/pinocchio/devel/coverage/.

Can you update your PR to reflect this observation?

nim65s, 2018-10-05:
that's true, gcovr's man page was misleading… I'm on it :)

jcarpent, 2018-10-05:
Thank you very much for this fix.

commit 0bd29d59 (2018-10-05): [Coverage] Temporarly exclude bindings/python directory from coverage reports, fix #534
commit f606f337 (2018-10-05): Merge pull request #538 from nim65s/devel

 [Coverage] Temporarly exclude bindings/python directory from coverage reports, fix #534
commit 9a0eb884 (2018-10-24): [Coverage] Temporarly exclude bindings/python directory from coverage reports, fix #534
commit edf56756 (2018-10-24): Merge pull request #538 from nim65s/devel

 [Coverage] Temporarly exclude bindings/python directory from coverage reports, fix #534
closed 2018-10-05 by jcarpent