stack-of-tasks/pinocchio #545: [jointJacobian] different behavior between C++ and python

gabrielebndn, 2018-10-16:
Python method `jointJacobian` DOES NOT call its corresponding C++ function. Instead, it internally calls `computeJointJacobians` followed by `getJointJacobian`. This causes three differences in the behavior:
- first, the unimportant, obvious ones: at the end of the python method, the overall Jacobian is computed, while at the end of the C++ method it is not. Also, the python method defeats the purpose of being more efficient, but who cares about efficiency in python
- second: the python method allows to specify a Reference frame (LOCAL/WORLD), while the C++ method does not and only offers the local version.
- third: `computeJointJacobians` internally does the equivalent of `forwardKinematics`, which C++ `jointJacobian` does not. This is particularly bad, because the doxygen C++ comment to `computeJointJacobians(model, data, q)` explicitly states:
> This function computes also the forwardKinematics of the model.
while the one for `jointJacobian` says:
> This function is equivalent to call first computeJacobians(model,data,q) and then call getJointJacobian<LOCAL>(model,data,jointId,J).
but this is not true. If you think this is not relevant, well, it was causing problems in one example Nicolas prepared in PR #541.
Should this be fixed? In which way? Or, should we leave things as they are in order not to break everybody's code, but explicitly document the difference in behavior?

gabrielebndn, 2018-10-16:
Update: actually, python's `jointJacobian` behavior is still different than that: it takes a boolean `update_kinematics` parameter (for which no default value is provided) and if this is true, then `computeJointJacobians`, otherwise not, and  and then `getJointJacobian` is called. This is explicitly documented in the python docstring which says (true = update the value of the total jacobian).

So, basically, the C++ and the python version are two different methods, which were thought to operate and be employed differently from each other. The python version is a convenience method which allows to follow the usual `computeJointJacobians` + one ore more `getJointJacobian` procedure while skipping the explicit call to `computeJointJacobians`, while the C++ version is a truly different algorithm which is made for one-off use.

I don't know, should I make the python behavior identical to the C++ one and maybe add an option to C++ to include kinematics computation, or leave things as they are and just be really clear in the documentation about the different behavior?

jcarpent, 2018-10-16:
I don't have time to investigate further, but the main idea is that Python follows C++, not the reverse.

jcarpent, 2018-10-16:
But, I would suggest to let open this issue and to solve it after the release.

nmansard, 2018-10-16:
The main 2 functions should be computeJointJacobians (compute data.J from root to leaves) and getJointJacobian (get some columns of data.J). These 2 should be bind in Python. 

The secondary function should be jointJacobian --possibly with another name-- that compute a single jacobian from leaf to root. It is more user friendly. Two solutions might be proposed to implement it: either compute the jacobian in data.J and return a reference to it (rnea style) ; or let the user allocates its own matrix J and compute in it. The first solution is bad because data.J should be used for computeJointJacobians ; the second is bad because it is asking the user to allocate memory, which is not friendly and not following the style of Pinocchio. I do not have a good solution.

Then Python API should follow C++ API.

gabrielebndn, 2018-10-16:
I would not say asking the user to allocate memory is not following the style of Pinocchio: that's exactly what getJointJacobian does.
Writing the result in data.J (which corresponds to the behavior of a deprecated method) is particularly bad, since `data.J` would represent different things depending on what methods are called. I'm not only talking about the overall Jacobian vs specific Jacobians, but also of specific Jacobians vs each other. Consider what would happen in the following (buggy) case:
```
se3::Data::Matrix6x & J1 = se3::jointJacobian(model, data, q, jointId_1)
se3::Data::Matrix6x & J2 = se3::jointJacobian(model, data, q, jointId_2)
```
Ok, you should not call jointJacobian twice, and instead call computeJointJacobians first and then getJointJacobian twice, but choosing to write into data.J seems to encourage particularly error-prone code. The solution
```
se3::Data::Matrix6x J1 = se3::jointJacobian(model, data, q, jointId_1)
se3::Data::Matrix6x J2 = se3::jointJacobian(model, data, q, jointId_2)
```
is not good, because it will amount to a copy into J1 and J2, i.e. a hidden allocation.

I suggest to leave the C++ method as it is. Only, in the doxygen comment, it should be made clear that this function is NOT equivalent to  computeJointJacobians + getJointJacobian, and in particular forwardKinematics is NOT called.

Then the python method would be made identical to C++, with the only difference that the output matrix, instead of being given, would be created, set to zero and returned (I think this is acceptable in python). Its signature would be `jointJacobian(model,data,q,JOINT_ID)`. Therefore, the python function will lose the capability to return the Jacobian in the world frame. Also, it will then become necessary to explicitly call forwardKinematics if data.oMi is to be read, as it is in C++. 

Usage of the python function as `se3.jointJacobian(model,data,q,JOINT_ID,ReferenceFrame,update_kinematics)` should be kept for compatibility for a while, keeping the same behavior, but be deprecated. It will require some tweaking to deprecate the function only when it is called in a certain way but not in others, but it can certainly be done. An alternative is to rename jointJacobian (the current version in C++ and the new one in python) to something else, like `computeJointJacobian`.

If we want to wait for version 2.0 in order to implement these changes but also keep working on the doc in #541, then I suggest to employ the computeJointJacobians + getJointJacobian paradigm in the python examples too.

jcarpent, 2018-10-17:
The point is that algo in Pinocchio should not return their results but should take as input their returns.
For instance, `jointJacobian`or `computeJointJacobian`should take at the last argument `const Eigen::MatrixBase<Matrix6Like> & Jout`.

nmansard, 2018-10-17:
This makes the following approximate code:
MatrixXd J;
jointJacobian(model,data,q,J);
Sorry, but I prefer:
J = jointJacobian(model,data,q);
especially in Python. Let's say my taste is not relevant.

nim65s, 2018-10-17:
I also prefer the second version. 

In python, it is highly unconventional to update one of the input of a function, return nothing, and expect the user to understand what happened.

jcarpent, 2018-10-17:
But to be efficient, the first approach is expected.
New algorithms in Pinocchio have been developed following this paradigm.

For Python, you may have a different API for the return type, no matter the solution in C++.

gabrielebndn, 2018-10-17:
I agree with @jcarpent. In C++ the users gives the memory, but in python the matrix is returned. It is just a different way of returning the matrix.

As a compromise, we might have the C++ method zero out the provided matrix and return a reference to the same matrix which was passed as input. This way we could write
```
MatrixXd J(6,model.nv);
doStuff(jointJacobian(model,data,q,J));
```
which is more compact than
```
MatrixXd J(6,model.nv);
J.setZero();
jointJacobian(model,data,q,J);
doStuff(J);
```

nmansard, 2018-10-18:
How many other algorithms in Pinocchio have the same behavior? I guess all other derivative algorithms?
As Jacobian is a very standard algo, to be used by newcommers, what about adding another matrix field in data to avoid the preallocation of J by the user in the "more compact" form proposed by @gabrielebndn ?

gabrielebndn, 2018-10-18:
@nmansard I advise against it. At first I thought it could be a solution, but then I realise it has its own problems, because the same matrix would potentially represent different joint Jacobians and this in not good either. Again, consider the problems potential users could incur to if they did:
```
se3::Data::Matrix6x & J1 = se3::jointJacobian(model, data, q, jointId_1);
se3::Data::Matrix6x & J2 = se3::jointJacobian(model, data, q, jointId_2);
```
The only *clean* way to do so would be to add a whole new vector field containing one Jacobian for each joint, which would be a total waste of space (we should do the same for frameJacobian too).
My compromise suggestion was different: basically, the matrix would still be provided by the user, only pinocchio would set it to zero and return a reference to the same matrix which was passed as input, so to be able to use the expression in nice one-liners

jcarpent, 2018-10-18:
I agree with @gabrielebndn, the whole idea of Jacobian is initially to not allocate memory for each joint.
We cannot revert it because it is the correct things should go.
Adding new data.Jjoint would be inefficient in terms of memory footprints for instance when we need to have all the data in the L1 cache for instance.

The correct way to handle the sparsity through MotionSet. it is also what I suggested in https://github.com/stack-of-tasks/pinocchio/projects/2

gabrielebndn, 2018-10-18:
The same behavior of jointJacobian is followed by all other derivative algorithms, plus a couple of recent convenience methods called getFrameVelocity and getFrameAcceleration (but I think we could easily switch to a more user-friendly behaviour for those two, I will investigate this further).

jcarpent, 2018-10-18:
An easy solution is to rely on computeJointJacobians and getJointJacobian.

gabrielebndn, 2018-10-18:
@jcarpent that would not address Nicolas' comments, since getJointJacobian *also* asks the user to allocate for J. Indeed, I think that whatever we choose for jointJacobian w.r.t. the return policy, we should make the same in getJointJacobian, frameJacobian, and getFrameJacobian.
If you were referring to PR#540, then I agree, I think this discussion is blocking the progress, and to avoid any inconsistency we should use computeJointJacobians and getJointJacobian in the examples, until we have made a decision on this.
(BTW, frameJacobian is not provided in C++, it is only provided as a Python binding, where it is implemented in the same way as the one for jointJacobian. We should fix this, it's on the to-do list of the python bindings project. But we should only fix this *after* the decision is final on whether to modify jointJacobian and/or rename it).

jcarpent, 2018-10-18:
No, when you ask the user, you only require that the user allocates just needed memory for his stuff, not for all the joints.

gabrielebndn, 2018-10-18:
@jcarpent I am not following you. Right now, jointJacobian and getJointJacobian both ask the user to allocate a single J, where the result will be put. They are identical w.r.t. that aspect, as for now.

jcarpent, 2018-10-18:
If the user calls several times jointJacobian for the same joint, then you will have at each call memory allocation. While this is not true when first allocating, then calling getJointJacobian

gabrielebndn, 2018-10-18:
What? I do no think this is the case right now. The following two pieces of code should make the same number of memory allocations:
```
se3::Data::Matrix6x J(6,model.nv); J.fill(0);
for (int i=0 ; i<100 ; ++i) {
      se3::computeJointJacobians(model,data,q);
      se3::getJointJacobian<se3::LOCAL>(model,data,JOINT_ID,J);
}
```
```
se3::Data::Matrix6x J(6,model.nv); J.fill(0);
for (int i=0 ; i<100 ; ++i) {
      se3::jointJacobian(model,data,q,JOINT_ID,J);
}
```

jcarpent, 2018-10-18:
Yes, but not the one suggested 'J = se3::jointJacobian(model,data,q,JOINT_ID)'

gabrielebndn, 2018-10-18:
Exactly. That's why I think we should not do that.

jcarpent, 2019-02-13:
@gabrielebndn Ping. What is the status of this issue?

gabrielebndn, 2019-02-13:
We haven't talked about it, but it's still in my mental list of TODOs. I am reluctant to take action because I feel any change might potentially change some behavior and I feel any decision on this should be shared. If you agree, I will leave C++ untouched and I will change Python's jointJacobian to reflect C++ . Then I will probably take similar measures with the frameJacobian

jcarpent, 2019-02-13:
I do agree. You can then move forward.

gabrielebndn, 2019-02-13:
Will do

jcarpent, 2019-02-28:
Solved by #698 .

PR #698 (2019-02-27): Align Python Jacobian to C++, fix #545
Mostly four things:
- Add a benchmark called `timings-jacobian` to compare between `getJointJacobian`, `computeJointJacobians` and related algorithms.
- Use the corresponding C++ function for the Python bindings of `jointJacobian`. Now, the signature and side-effects of `jointJacobian` in Python are completely aligned to C++. The old signature is now deprecated. This fixes #545
- Implemented missing `frameJacobian` function in C++, and use the newly-created function for the Python bindings of `frameJacobian`. The old signature is now deprecated. This also [... cut]
commit 9f7aba29 (2019-02-27): Merge pull request #698 from gabrielebndn/topic/jacobians

Align Python Jacobian to C++, fix #545
closed 2019-02-28 by jcarpent