stack-of-tasks/pinocchio #547: Pinocchio 2.0

nmansard, 2018-10-17:
@jcarpent is preparing a major change of API. Here are some suggestions to prepare the release.
- initial candidate release should be proposed as 1.99.x
- propose a guideline to convert developments based on 1.3 to 2.0
- don't move master to a 2.0 version until the main dependencies of Pinocchio have given their agreement.
- during the transition phase, pinocchio/devel will be in 2.0, while pinocchio/master will be in 1.3. Bug fixes can be pushed to pinocchio/master during the transition phases. Parallel developments should enforce that the same bug fixes are pushed in pinocchio/devel. 
- as the master branch will diverge from the devel branch, we will have to revert possible commits in master before merging it to devel.
- we shoul wait for a consensus among the users of Pinocchio before merging master to devel.

It is suggested to developers using Pinocchio to start moving their devel branch to API 2.0 when it is officially released. If possible, we would like to set up a dead line for the transition phase. Pinocchio/master will be merged to version 2.0 after this deadline.

In summary, here would be the agenda:
0. update pinocchio/master to the latest commit.
1. merge new API in pinocchio/devel and tag it 1.99.x
2.a validate pinocchio/devel and release it as version 2.0.0 when solid enough.
2.b during this period, master is staying in 1.3.x but can be added with bug fixes
3. once 2.0 is officially released, the software depending on pinocchio should move their devel branches to pinocchio/devel
4. at a given date (consensually defined), pinocchio/master is merged to pinocchio/devel (after reverting possible head commits). Pinocchio 2.0 becomes the official version.

jcarpent, 2018-10-17:
From my side, I will merge the current devel branch into 2.0.0 to have an aligned history.
It would be nice to not modify the current devel branch in order to not add more stuff in my todo list.
The PR can remain open and would be merged in the new version by their authors.

nim65s, 2018-10-17:
I can take care of the robotpkg part

nmansard, 2018-10-17:
When merging devel, please follow the guidelines I described upper (or discuss it beforehand).

jcarpent, 2018-10-17:
Yes, this is pretty standard. No worry!

jcarpent, 2018-10-17:
By the way, I will release a new version of Pinocchio before proposing 2.0.0-beta.
This will add the last fixes already present in the current devel branch.

nmansard, 2018-10-18:
It was propose to release in 1.99.0 for the candidate.

jcarpent, 2018-10-18:
1.99.0 or 2.0.0-alpha makes no difference in terms of semantics, but it does for the checking of the version.
When using the macro PINOCCHIO_AT_LEAST_VERSION(2,0,0), which what we have to use to ensure at compile time the current version of Pinocchio, is the correct of ensuring that we are the correct version, not PINOCCHIO_AT_LEAST_VERSION(1,99,0).
By the way, this is also how Eigen works.

So I will use 2.0.0-alpha for the pre-release version.

aelkhour, 2018-10-24:
Is there currently a target date for this release?

jcarpent, 2018-10-24:
I hope by the end of the next week.

jcarpent, 2018-10-29:
As release 1.3.2 is buggy, I will make a new one called 1.3.3 with the fix.
Thanks to @aelkhour for raising this issue.

jcarpent, 2018-10-29:
I've just opened a PR #569.
This PR should normally compile on Mac OS X and ubuntu 14.04. I did try on other systems yet.

I will test it on the Gitlab of LAAS.

I hope that all the algorithms will continue to work with most of the projects relying on Pinocchio.

jcarpent, 2018-10-29:
I've just merged #569.
I also added the tag `2.0.0-alpha` for this pre-release version

gabrielebndn, 2018-10-29:
I confirm it is building on ubuntu 16.04

closed 2019-01-11 by jcarpent