stack-of-tasks/pinocchio #548: Geometry objects in python (and connection with Gepetto-viewer)

nmansard, 2018-10-17:
I am currently working with models that have no meshes but only simple geometries (capsule). As far as I know, simple geometries are not bind to python (yet). Could anybody tell me what is the current status of this development, and if anybody as some on-going development that I could finalize and merge? 

Once I have simple geometries in Python, I guess it would be nice to have them displayed in Gepetto-viewer. Did anybody worked on this?

Thx

jcarpent, 2018-10-17:
We already discussed this point a while ago.
We should handle the simple geometries through the FCL bindings.
I have already binded some of them, but it would be great to directly deal with the official Python bindings of FCL for compatibility reasons.

For the viewer part, it should be easy to do that.
But the main question is: do you we want to rely on gepetto-gui? My guess is no. And we should move to a new renderer (more flexible) such as Panda3d. See #180 for further details.

nmansard, 2018-10-17:
Do you have the pointers for the bindings of FCL and the work you started to do on bindings the geoms?

Let's keep the discussion about GepettoViewer out of this. Once I have the geoms in Python, forwarding them to GepettoViewer or any other viewer would be simple work.

jcarpent, 2018-10-17:
It is in a local repository. I will try to extract it from there asap.

cmastalli, 2018-10-18:
Guys this is another solution that I found, it works on Linux, Mac OSX and Windows:
https://github.com/OTL/urdf-viz

nim65s, 2018-10-18:
This depends on the https://github.com/sebcrozet/kiss3d 3d engine and the https://github.com/OTL/k kinematics library. I think we don't want to depend on a kinematics library. But thanks for the link, I will add it in pinocchio-benchmarks

nmansard, 2018-11-05:
@jcarpent did you manage to find your early work on FCL bindings?

jcarpent, 2018-11-05:
Not at all. Sorry about that.

closed 2019-02-19 by jcarpent