stack-of-tasks/pinocchio #549: Switch to BSD v2.0

olivier-stasse, 2018-10-17:
Dear all,
We will soon switch to BSD License v2.0 to ease our collaborations.
Let me know if there is a problem with this.
Cheers,
Olivier.

jcarpent, 2018-10-17:
This has to be discussed with all the partners of the project: Wandercraft, INRIA, ... and also the Authors.
From my side, this change of license has to be discussed and justified further.

jcarpent, 2018-10-18:
@olivier-stasse The first package that should be modified for the license is the jrl-cmake-modules, which is LGPL.
Please consider this package as your first priority.

nmansard, 2018-10-18:
I will contact WAN to check. I don't think it is a problem.
Sorry, but INRIA is not yet a partner of this project. We can revert any commit by INRIA if you believe that some of your recent commit is more than just release your previous work done when under CNRS contract.

jcarpent, 2018-10-18:
The code I'm writing now (with new features) and I'm pushing now is INRIA.

jcarpent, 2018-10-18:
Just to be clear, release 2.0.0 has been made (also) during my stay at INRIA, under the LGPL license. I won't revert any commit from it as I'm about to release my code.

aelkhour, 2018-10-18:
> I will contact WAN to check. I don't think it is a problem.

@nmansard @olivier-stasse Wandercraft agrees with switching to the standard three-clause BSD license (BSD License v2.0).

jcarpent, 2018-10-18:
@aelkhour Thanks for your answer.

jcarpent, 2018-10-18:
@aelkhour The same applies for eigenpy. Does Wandercraft agree?

aelkhour, 2018-10-18:
> @aelkhour The same applies for eigenpy. Does Wandercraft agree?

Yes Wandercraft also agree to switch eigenpy to the standard three-clause BSD license (BSD License v2.0).

I'm not sure it's the case for eigenpy, but Wandercraft has authored a number of commits in pinocchio. @olivier-stasse Could you ensure that Wandercraft still appears as author after making the license switch ?

olivier-stasse, 2018-10-18:
@aelkhour. We will pay attention with respect to eigenpy.
@jcarpentier. LGPL is problematic for code which are embedded. This has
happen for a lot of projects where embedded application are being
developped.
The code is statically embedded in the application and thus become
contaminent.
I think this is not nice for our industrial partner.

The LGPL licence has further problem: most of our code is not
industrialized and very often need investment to be robustify and
integrated into real world application.
The BSD-clause 2 allows *ANYONE* to use for its own product, line etc... We
save time and money for everyone.

This is working quite well for the ROS community where companies are
investigating back to the community.
I intend to propose and push the same license switch with the
jrl-cmake-modules.

olivier-stasse, 2018-10-18:
One important detail after discussion with our legal department:
* We do not need to put the license on each of the files as long as it is on the main repo

jcarpent, 2018-10-18:
@olivier-stasse Thanks a lot for these details that should be available for Pinocchio users to clearly understand the change of license. (I mean this is not obvious why the changing of license is so important).

jcarpent, 2018-10-24:
Please wait for release 2.0.0 to proceed to the change of license.

nmansard, 2018-10-26:
INRIA confirms that we do not have to ask their permission if changing license of 1.3.
PAL confirms that BSD is better for them.
As all parties involved in 1.3 agree, I think we should change it now. This does not impact the code, so no need to synchronize with v2.0.

jcarpent, 2018-10-26:
I'm almost done with Pinocchio 2.0.0-alpha.
The merging was a painful task. I propose to change the license just on top of it (it will change all the headers by the way).

jcarpent, 2018-10-26:
By the way, I will do it by myself.

jcarpent, 2018-11-09:
Fixed by #592.

closed 2018-11-09 by jcarpent