stack-of-tasks/pinocchio #551: [Versioning] Pinocchio still thinks it is version 1.2.9

gabrielebndn, 2018-10-17:
Macros
    PINOCCHIO_MAJOR_VERSION
    PINOCCHIO_MINOR_VERSION
    PINOCCHIO_PATCH_VERSION
were not updated correctly since we switched to 1.3.0.
We are at 1.3.1 now, but Pinocchio still returns 1.2.9.

How should we handle this?
Is there a way to back-fix this problem in the previous two releases?
Should we correct to 1.3.1 immediately?
Should we leave it as it is and only correct in the next release, the one just before 2.0?

jcarpent, 2018-10-18:
I have a fixe for this which is for the next release. This was due to the cmake module which was not able to handle it.
Unfortunately, there is no fix for the previous releases. I will just to make a patch for the current devel branch asap.

gabrielebndn, 2018-10-18:
Ok, I also think we should wait for a new release to correct the problem, it does not make sense to change those values between two different releases, it would only make the problem worse.

So, I did not understand, for correcting the release number we wait, but should I fix the binding for `printVersion()`? In python, `se3.printVersion('.')` works, but `printVersion()` does not: it was returning garbage on my Macbook, and it's causing a segfault (in Python!) on ubuntu. The C++ version is fine, except that the value is wrong of course. It will be super quick.

One consideration out of this: should we keep a record of known (and hopefully fixed) bugs for each version? Maybe a changelog?

gabrielebndn, 2018-10-18:
Looks cool. Thanks

gabrielebndn, 2018-10-29:
Hate to raise this, but after the version macros were correctly updated to 1.3.2, now that we switched to 1.3.3 they still say 1.3.2...

jcarpent, 2018-10-29:
Yes, you're right. I forgot to do it manually.
I think this is not dramatic for this release.

And in the future, this will be handled automatically thanks to https://github.com/jrl-umi3218/jrl-cmakemodules#162

gabrielebndn, 2018-10-29:
Yes, it is not dramatic, it's just an additional patch. I just wrote it to let it be known.
Looking forward for the automated solution.

jcarpent, 2018-10-29:
This issue has been solved by #569.

closed 2018-10-29 by jcarpent