stack-of-tasks/pinocchio #553: Contributor list

nmansard, 2018-10-19:
Currently, contributors are formally listed in the README.md as Carpentier, Valenza, Mansard et al. We are now thinking of transferring or duplicating this info in the documentation, hence the opportunity to update it. I do not think that the initial developments by @fvalenza make him in the top 3 dev of Pinocchio. We might either decide of the top 3, or find a way of more generally and versatility acknowledge for the contributions. 

I suggest to have a page listing the contributors, possibly in the documentation, and refer to the general citation as:
@misc{pinocchioweb,
      author = {"Pinocchio development team" and "(see https://stack-of-tasks.github.io/pinocchio/contributors)"},
      title = {Pinocchio: fast forward and inverse dynamics for poly-articulated systems},
      howpublished = {https://stack-of-tasks.github.io/pinocchio},
      year = {2015--2018}
     }

jcarpent, 2018-10-19:
A clear contribution list is available here:
https://github.com/stack-of-tasks/pinocchio/graphs/contributors
I think the idea of Eigen contributors is good: the name of the main (or original) developers + and others (similar to Pinocchio development team).
The idea of having the name of the main developers is really for academic, hiring reasons and for recognition in the community mostly.

jmirabel, 2018-10-19:
I don't think you can trust the graph provided by git/github to have a relevant contribution list. The list you are talking about includes correction of types, re-indention, dummy changes... I personally don't trust it.

I agree Eigen's method is better although it has to be manually maintained (so one has to think of doing this).

jcarpent, 2018-10-19:
Yes, I agree. The Github one will artificially increase the contributions. But, it has the advantage of listing the contributors in an automatic way, that was my point.

nmansard, 2018-10-19:
So @jcarpent you are suggesting we keep 3 names in the author list of the bib ref, and chose the 3 names as the automatic top 3 contributors. I dont care as long as we have a policy decided. But I think this policy is not very fair, only automatic.

jcarpent, 2018-10-19:
From my observation as the main maintainer of the project, the main contributors in Pinocchio I see are:
- Joseph for the Lie group support among others
- Nicolas for the implementation of the first features among others
- Justin for the core developments among others

Florian was originally a core member, but the project has also evolved a lot since then. So his name should appear at least in the credits, but maybe not in the main authors' list, but I'm not able to evaluate this point alone.

I would suggest something like that:
author = Carpentier, Mirabel, Mansard and the Pinocchio development team

If someone feels missing, can he add his feedback to the discussion?

gabrielebndn, 2018-10-19:
I'm only writing Pinocchio development team for the time being. Feel free to change it later.

jcarpent, 2018-10-19:
You can take this one I guess is much more closer to the original one which is already used by the way: Carpentier, Mirabel, Mansard and the Pinocchio development team

jcarpent, 2018-10-19:
Just for information, this is the way that drake (https://drake.mit.edu/) from MIT is cited:
```
@misc{drake,
 author = "Russ Tedrake and the Drake Development Team",
 title = "Drake: A planning, control, and analysis toolbox for nonlinear dynamical systems",
 year = 2016,
 url = "https://drake.mit.edu"
}
```

jcarpent, 2018-10-19:
And the title should be something like:
**Pinocchio: a fast and flexible implementation of Rigid Body Dynamics algorithms and their analytical derivatives**

gabrielebndn, 2018-10-19:
I tend to agree with @jcarpent about the author list, by think this should be done in agreement with everybody and I do not want to push any particular view.
As for the title proposed, I think it is way too long. I would keep it simple, like "Pinocchio: Rigid Body Dynamics Algorithms [and their Derivatives]", but whatever.
But I know one thing: whatever choice is made, it should be reflected as the subtitle appearing in the main page https://stack-of-tasks.github.io/pinocchio/. Which, by the way, contains a typo right now (it should be dynamics, not dynamic)

jcarpent, 2018-10-19:
Yes, the website is not really up to date and should be modernized after the documentation is done.
I think the mention to derivatives is important. We should keep it!

nmansard, 2018-10-19:
If you have to select three names, let's choose the three followings. Just keep in mind that we may have to revise that in the future, and that it might become an unpleasant discussion.

There are many features in Pinocchio, so I don't think we should focus the title on the derivatives, also it is true it is a good selling argument as it is quite a unique feature. But let's use it as one of the main argument, and keep the title clean. I personally like clean and short titles. What about the following reference?

I agree with @gabrielebndn that the title should be use in the main page of Pinocchio.

@misc{pinocchioweb,
author = {Justin Carpentier and Joseph Mirabel and Nicolas Mansard and "the Pinocchio development team"},
title = {Pinocchio: fast forward and inverse dynamics for articulated systems},
howpublished = {https://stack-of-tasks.github.io/pinocchio},
year = {2015--2018}
}

jcarpent, 2018-10-19:
I don't think that we have to restrict Pinocchio to forward and inverse dynamics (this is very suited for the SoT for instance). Pinocchio goes beyond and it is very related to rigid body dynamics algorithms.

jmirabel, 2018-10-19:
I would go for
- *Pinocchio: Rigid Body Dynamics algorithms*,
- *Pinocchio: Rigid Body Dynamics for articulated systems* (my favorite because I do not think the word algorithm adds anything).

gabrielebndn, 2018-10-19:
In my view, "Rigid body dynamics algorithms" *includes* the derivatives, but it is true that the derivatives are a unique feature of Pinocchio and thus a good selling point. I like @jmirabel's last proposal

gabrielebndn, 2018-10-19:
Going for "Justin Carpentier and Joseph Mirabel and Nicolas Mansard and the Pinocchio Development Team", I'll let you guys modify the title

proyan, 2018-10-19:
>     * _Pinocchio: Rigid body dynamics for articulated systems_ (my favorite because I do not think the word _algorithm_ adds anything).

Shouldn't there be "efficient" there somewhere? Or the fact that it is templated?
Eigen's one line introduction reads: "Eigen is a C++ template library for linear algebra"

jmirabel, 2018-10-20:
Since we talk about title and important features, we can first list the features we want to put forward. As far as I know, the features are (with no specific order):
- Pinocchio is fast:
  - C++ template library,
  - cache friendly,
  - automatic code generation.
- Pinocchio implements rigid body dynamics algorithms:
  - forward and inverse dynamics,
  - analytical derivatives of inverse and forward dynamics.
- Pinocchio is multi-thread friendly.
- Pinocchio is reliable and extensively tested (unit-tests, simulations and real robotics applications).
- Pinocchio is cross-platform (windows ?)

Feel free to edit this list.

Now for the title, I am confused. I understand from the discussion that *rigid body dynamics* should be in the title. The choice of a title depends more on the desired future of Pinocchio than on our personal desire. Maybe *Pinocchio: efficient library for articulated rigid body dynamics* ?

proyan, 2018-10-23:
@jcarpent @nmansard 

While I agree that pinocchio has evolved a lot since @fvalenza's contributions, I see that his name has been removed even from the credits list. I am not sure what the policy regarding the credits is, but this seems to be a mistake to me.

nim65s, 2018-10-23:
@proyan he has not been removed from this list ; this is just a new list that has been added [here](https://github.com/stack-of-tasks/pinocchio/commit/6be296ecac0a6e4cdc77451c8f8d958ebbbf8780#diff-04c6e90faac2675aa89e2176d2eec7d8R80), before this discussion started.

Now, we are discussing this list :)

jcarpent, 2018-10-23:
This is a mistake. I did not put all the names in the credit lists at all.

jcarpent, 2018-10-23:
@proyan  By the way, it is clearly written:
`If you have taken part to the development of **Pinocchio**, feel free to add your name and contribution here.`

jcarpent, 2018-10-23:
I've just added his name on the REAME.md. See 0ee62897b93fb4b30c3dde58ca75a74a83d1bbf6.

gabrielebndn, 2018-10-23:
@jcarpent, I see there's an additional commit you put in master but not in devel:  6bcdc2a (README: add Pinocchi features). Could you put that in devel too?

jcarpent, 2018-10-23:
@gabrielebndn: Done!

gabrielebndn, 2018-10-23:
By the way, I see the README only mentions the pinocchioweb entry. Shouldn't we also mention the SII article, as we do in the new Overview?
(For the time being we can directly write the bib entries into README.md because it seems there is a problem displaying files we \include on a browser)

jmirabel, 2018-10-23:
I see that someone has added the feature list I mentioned. When I said
- *Pinocchio is cross-platform*,

I think this is a bit overselling. I think we should say
- *Pinocchio is supported on MacOs and many Linux distribution ([See build status here](http://robotpkg.openrobots.org/rbulk/robotpkg/math/pinocchio/index.html))*

jcarpent, 2018-10-23:
@gabrielebndn As the paper is still under review, I would suggest to wait until the final version.
@jmirabel You're right. I'll modify it.

jcarpent, 2018-10-24:
I will close this issue, I think the issue has been solved. Feel free to reopen if needed.

closed 2018-10-24 by jcarpent