stack-of-tasks/pinocchio #556: Git tree

nmansard, 2018-10-24:
I am surprise by the current development of the repo tree.
- we had a fork between devel and master. Why is so? The policy of the project has always been to keep master behind devel. I think we should stick to this policy, and in case of real need, discuss it here beforehand. 
- we merge @nmansard/devel into topic/doc-v2 . @nmansard/devel was meant to be merged in devel , while topic/doc-v2 was supposed to be kept uptodate with devel by regularly merging devel in topic/doc-v2. Is that my mistake? Should I keep the idea of merging #554 in devel?

nmansard, 2018-10-24:
Wrong reading of the tree for the second point. Forget it.

jcarpent, 2018-10-24:
It is just at the README level. It because we have used the internal editor of GitHub to edit the README.md and using cherry-pick.
I have aligned both histories in https://github.com/stack-of-tasks/pinocchio/tree/devel-sync, that I can now push forward to devel if you agree on this point.

nmansard, 2018-10-24:
Ok.

closed 2018-10-24 by nmansard