stack-of-tasks/pinocchio #559: [math] Build error with SINCOS macro on Linux and float

aelkhour, 2018-10-25:
On Linux, the `SINCOS` macro simply forwards the call to the [sincos](http://man7.org/linux/man-pages/man3/sincos.3.html) function.

As the output arguments are pointers, `sincos` only compiles for `double` but not `float`. This renders explog functions unusable with scalars other that `double`.

I don't know what a proper fix would be in this case.

jcarpent, 2018-10-25:
I will provide one in version 2.0.0. 
But maybe, you need a fix now?

By the way, why float and not double?

aelkhour, 2018-10-25:
Thanks @jcarpent, it's not urgent so we can wait until v2.0.0.

We internally use pinocchio with float, never with double. I don't have a proper justification for that, other than performance issues. We did not benchmark this choice, so I can't exclude that we will revert back to using double some day.

olivier-stasse, 2018-10-25:
My own experience is that computation are *really* less precise/stable with float than double.
If your CPU can handle directly 64 bits I would suggest to stick with it.
If not then indeed you are on a very constrained environment.
Then you probably would benchmark pinocchio more closely. The prediction capabilities of your CPU can have an impact.

jcarpent, 2018-10-25:
It should be fixed by #563. 
@aelkhour Can you try it?

closed 2018-10-27 by jcarpent