stack-of-tasks/pinocchio #562: [spatial] Symmetric3Tpl::rotate triggers assert with float in some cases

aelkhour, 2018-10-25:
`Symmetric3Tpl::rotate` contains an assert which relies on [isApprox](https://eigen.tuxfamily.org/dox/classEigen_1_1DenseBase.html#title47). When using `float` scalars, this assert is sometimes triggered in Debug as the precision argument is kept to its default value.

IMHO it would make sense to rely on something like `std::numeric_limits<_Scalar>::epsilon()` and pass it to `isApprox`.

jcarpent, 2018-10-25:
`std::numeric_limits<_Scalar>::epsilon()` is sensitive to rounding errors for two pipelines of computations, while `Eigen::NumTraits<Scalar>::dummy_precision() ` will lower this precision.

In Eigen, this is the value of the dummy_precision for a float:
```
template<> struct NumTraits<float>
  : GenericNumTraits<float>
{
  EIGEN_DEVICE_FUNC
  static inline float dummy_precision() { return 1e-5f; }
};
```
for a precision of a float of 5.96e-08, which seems reasonable.

Can you output the results of `(R.transpose()*R)` when it occurs?

matthieuvigne, 2018-10-25:
In fact, the problem comes from the fact that we typically mix float and doubles in our code.

The following code will trigger the assert

```c++

#include <iostream>
#include <iomanip>

#include <pinocchio/spatial/inertia.hpp>
#include <pinocchio/spatial/fwd.hpp>

int main(int argc, char **argv)
{
	// Create inertia object.
	se3::Inertia i = se3::Inertia::Zero();

	// Create a random quaternion.
	Eigen::Quaternion<float> quat(1.0, 1.0, 0.0, 0.0);
	quat.normalize();

	std::cout << std::setprecision(10);
	std::cout << "Quaterion:" << quat.coeffs().transpose() << std::endl;
	std::cout << "Quaterion norm:" << quat.norm() << std::endl;
	std::cout << "R.T R" << std::endl;
	std::cout << quat.matrix().transpose() *  quat.matrix() << std::endl;

	// Try to rotate symetric3 object.
	i.inertia().rotate(quat.matrix().cast<double>());
	return 0;
}
```
Output:
```
Quaterion:0.7071067691            0            0 0.7071067691
Quaterion norm:0.9999999404
R R.T
           1            0            0
           0 0.9999998808            0
           0            0 0.9999998808
Assertion `(R.transpose()*R).isApprox(Matrix3::Identity())' failed.
```

However, this comes from the fact that the assert is done with double precision on a matrix casted from float (dummy_precision for double is 1e-12). If we were to do this with a float inertia object, things would work fine.

jcarpent, 2018-10-25:
You can use for your computations something like:
```
typedef se3::InertiaTpl<float> Inertiaf;
```
and do all your computations with it.
But, this error is expected if you mixed both float and double.

One solution we would be to select the highest dummy_precision() value from the two Scalar types.
This feature has then to be spaned in the whole code.

matthieuvigne, 2018-10-25:
For our own purpose, we will as you said set this assert to work with float precision on doubles. But I don't think this change is required for pinocchio itself - working with a higher than normal precision means we could miss some numerical errors when working with doubles.  pinocchio in itself works fine on both double and float.
I think we can close this issue, @aelkhour do you agree?