stack-of-tasks/pinocchio #566: Build error related to fcl on pinocchio v1.3.2

aelkhour, 2018-10-26:
While trying to build pinocchio v1.3.2, I encountered this build error:

```
[  3%] Building CXX object src/CMakeFiles/pinocchio.dir/parsers/sample-models.cpp.o
/usr/bin/g++   -DWITH_URDFDOM -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion -fPIC -O3 -DNDEBUG   -o sample-models.cpp.o -c src/parsers/sample-models.cpp
sample-models.cpp: In function ‘void se3::buildModels::addManipulatorGeometries(const se3::Model&, se3::GeometryModel&, const string&)’:
sample-models.cpp:198:53: error: ‘Sphere’ is not a member of ‘se3::fcl’
                                   boost::shared_ptr<fcl::Sphere>(new fcl::Sphere(0.05)),
```

Is this the expected behavior when hpp-fcl (optional dependency) is not installed ?

jcarpent, 2018-10-26:
Not expected at all. The addGeometry module should be dependent of HPP_FCL.
Thanks for raising this bug!

gabrielebndn, 2018-10-26:
In the future, to avoid this kind of problems, maybe automated tests should also be run for build configurations where optional dependencies are not installed. @nim65s, what do you think?

nim65s, 2018-10-27:
I think that at some point I already did that, but we changed our minds and I removed it.
I also think that we already run 10 tests that takes up to more than 2h.
And I also think that nobody run those tests anyway: https://gepgitlab.laas.fr/stack-of-tasks/pinocchio/pipelines the last one was 3 weeks ago.

jcarpent, 2018-10-29:
#568 has normally fixed the issue. @aelkhour Can you check it?

closed 2018-10-29 by jcarpent

aelkhour, 2018-10-29:
@jcarpent Tested, it builds now :+1:

jcarpent, 2018-10-29:
OK. Very nice.
I will make a new release now.
Thanks again for reporting the bug.