stack-of-tasks/pinocchio #572: PROJECT_CUSTOM_HEADER_EXTENSION change

nim65s, 2018-10-30:
Hi @jcarpent,

Could you explain why you changed the header extension in https://github.com/stack-of-tasks/pinocchio/commit/751d4bcc42495b48d04f2ba11efd3b2c0f313d62 ?

This breaks robotpkg's detection of pinocchio version.

jcarpent, 2018-10-30:
It should not break. I just mentioned that the default extension if .hpp and no more .hh.
Can you explain the impact?
How does it work with robotpkg?

nim65s, 2018-10-30:
To check if a package is installed and which is is current version, one can teach to robotpkg to look into a certain file for a certain string.

For [pinocchio](https://github.com/nim65s/robotpkg/blob/master/math/pinocchio/depend.mk), it is: 
```
SYSTEM_SEARCH.pinocchio=\
  'include/pinocchio/config.hh:/PINOCCHIO_VERSION/s/[^0-9.]//gp'	\
  'lib/libpinocchio.so'							\
  'lib/pkgconfig/pinocchio.pc:/Version/s/[^0-9.]//gp'
```

So we are looking for a line with the string `PINOCCHIO_VERSION` in a file called `include/pinocchio/config.hh`

Then, if we change this .hh suffix to a .hpp, robotpkg will assume that pinocchio is not installed, as it will not be able to find the `config.hh` file.

I think that we can update this detection rule to find a .hpp, but we should only to this if we had a reason, as this might mislead users without an up-to-date robotpkg repository. In other words, I would like to follow the general rule «[If it ain't broke, don't fix it](https://en.wikipedia.org/wiki/Wikipedia:If_it_ain%27t_broke,_don%27t_fix_it)»

closed 2018-10-30 by jcarpent

nim65s, 2018-10-30:
So you believe that it is better to change the "default behavior of cmake forcing the user to make use of *.hh"  into a "default behavior of pinocchio forcing the user to make use of *.hpp even if it was *.hh before, and this change will probably break a few things" ?

What is the mistake in using a .hh ?

I really don't get it, but it's up to you.

jcarpent, 2018-10-30:
The idea is to have all the headers in Pinocchio in .hpp and .hxx.
Choosing .hh instead of .hpp is another choice. But the JRL cmake module previously forced to have {config,warning, etc}.hh

jmirabel, 2018-10-31:
When you think renaming a file or a directory is a good idea, then, I think it is good practice to provide deprecation warning.
For instance, when renaming config.hh into config.hpp, add a file config.hh with
```cpp
#ifndef CONFIG_HH
#define CONFIG_HH

# warning "This file is deprecated. Include <config.hpp> instead"

# include <config.hh>

#endif

nim65s, 2018-10-31:
@jmirabel in this particular case, this will not prevent some people to ask me why they have an error message saying that pinocchio is not installed, even if they lost hours recompiling and reinstalling it, while they are trying to install py-pinocchio, sot-pinocchio, or hpp-pinocchio, but yes, it would still be a bit better :)

jcarpent, 2018-10-31:
@jmirabel This will be indeed a good way!

jmirabel, 2018-10-31:
Yes, I know it won't fix robotpkg.

I merely mentioned something that I think is a good general rule for backward compatibility.

jcarpent, 2018-10-31:
@nim65s Why `'lib/pkgconfig/pinocchio.pc:/Version/s/[^0-9.]//gp'`is not working as a substitute? It is weard!

jmirabel, 2018-10-31:
Isn't it a AND operation rather than a OR ?

nim65s, 2018-10-31:
I opened this error because it was not working, so I think this is an AND

jmirabel, 2018-10-31:
I think you should simply use this.
```
SYSTEM_SEARCH.pinocchio=\
  'lib/libpinocchio.so'							\
  'lib/pkgconfig/pinocchio.pc:/Version/s/[^0-9.]//gp'
```

jcarpent, 2018-10-31:
I do agree!

nim65s, 2018-10-31:
I was going for https://github.com/nim65s/robotpkg/commit/3975805ac0f404093e502295d863328d527daca8 as I was hoping this will not be changing again too much… But your right, it will be more wise to just remove it.