stack-of-tasks/pinocchio #577: Cannot build pinocchio v2.0.0-alpha from tarball

aelkhour, 2018-10-31:
pinocchio v2.0.0-alpha builds correctly when the package is cloned.

However, when I try to build it from the release tarball, since the `.version` file is added, it seems that `cmake/version.cmake` does not compute the major, minor and patch versions. As a result the beginning of the `config.hh` looks like this

```c++
#ifndef PINOCCHIO_CONFIG_HH
# define PINOCCHIO_CONFIG_HH

// Package version (header).                                                                                                                                                                                       
# define PINOCCHIO_VERSION "2.0.0-alpha"
# define PINOCCHIO_MAJOR_VERSION 
# define PINOCCHIO_MINOR_VERSION 
# define PINOCCHIO_PATCH_VERSION 
```
and `bindings/python/utils/version.cpp` fails to compile.

This might need a fix at the level of the cmake submodule if I understood correctly.

jcarpent, 2018-10-31:
Yes, it is indeed coming from the submodule.
Thanks for reporting this bug. I will fix it.

jcarpent, 2018-11-02:
@aelkhour I've just updated the tar.gz of the version 2.0.0-alpha. Can you check that everything works fine from your side?

aelkhour, 2018-11-02:
@jcarpent it works now, thanks! You can close this issue.

closed 2018-11-02 by jcarpent