stack-of-tasks/pinocchio #579: Add option to choose to build static or shared library

aelkhour, 2018-11-02:
In some cases it can be handy to generate a static library instead of a shared one.

Do you think such a patch could be integrated?

```patch
diff -r -u CMakeLists.txt CMakeLists.txt
--- CMakeLists.txt	2017-05-29 11:56:41.806068622 +0200
+++ CMakeLists.txt	2017-05-29 13:39:23.258088413 +0200
@@ -25,6 +25,7 @@
 SET(PROJECT_NAME pinocchio)
 SET(PROJECT_DESCRIPTION "Rigid multi body dynamics algorithms")
 SET(PROJECT_URL "http://github.com/stack-of-tasks/pinocchio")
+OPTION(BUILD_SHARED_LIBS "Build package with shared libraries." ON)
 SET(PROJECT_CUSTOM_HEADER_EXTENSION "hpp")
 OPTION(INSTALL_DOCUMENTATION "Generate and install the documentation" ON)
 
diff -r -u src/CMakeLists.txt src/CMakeLists.txt
--- src/CMakeLists.txt	2017-05-29 11:56:41.810068622 +0200
+++ src/CMakeLists.txt	2017-05-29 12:09:03.090071003 +0200
@@ -74,8 +74,8 @@
 )
 
 IF(UNIX)
-    # Create target libpinocchio.so
-    ADD_LIBRARY(${PROJECT_NAME} SHARED ${HEADERS} ${${PROJECT_NAME}_SOURCES})
+    # Create target libpinocchio.
+    ADD_LIBRARY(${PROJECT_NAME} ${HEADERS} ${${PROJECT_NAME}_SOURCES})
     SET_TARGET_PROPERTIES(${PROJECT_NAME} PROPERTIES LINKER_LANGUAGE CXX)
     IF(BUILD_WITH_COMMIT_VERSION)
       TAG_LIBRARY_VERSION(${PROJECT_NAME})
```

jcarpent, 2018-11-02:
Release 2.0.0 of Pinocchio will not have any shared library, because we aim at making Pinocchio a header only library.

In the meantime, we also want to give access to the user to a compiled library with default scalar being double. We are discussing this point in #571. You are certainly very interested in this feature?

aelkhour, 2018-11-02:
As we need to cross-compile pinocchio anyway, I'm not sure a pre-generated binary is of interest to us for the moment.

If pinocchio will be header-only, then I agree that there is no need for this option. Closing.

closed 2018-11-02 by aelkhour

jcarpent, 2018-11-02:
What do you need by cross-compiling?
We can think of making a new project on top of Pinocchio to facilitate the compatibility with your goals, and it will compile the library for a specific architecture.

jcarpent, 2018-11-02:
@aelkhour What do you think?

aelkhour, 2018-11-02:
For now we need to cross-compile our code for the QNX OS, using a proprietary compiler.

duburcqa, 2020-04-16:
This issue should be reopened since making Pinocchio a header only library is not longer expected in the future as far as a know.