stack-of-tasks/pinocchio #607: aba derivative returns different values for same states and controls

hammoudbilal, 2018-11-25:
I have been playing with the analytical derivatives on the devel branch for a while
I set q, dq, and tau to constant value, then I call computeABADerivatives multiple times for the same q, dq, and tau values. the resulting matrices start increasing in value for each time I call it. 
I tried following the unit tests but it seems that the data object is created separately for each case. 
I also tried setting data.ddq_dq, data.ddq_dv and data.Minv to zero ever time i call computeABADerivatives, but it didn't help. 
I am using the python binding to call the function, are there any other matrices that I need to reset that might be accumulating through the algorithm ?

gabrielebndn, 2018-11-25:
Hi @hammoudbilal, could you post a minimal example that reproduces your problem? Have you tried in C++? If yes, do you experience the same behavior?

jcarpent, 2018-11-25:
@hammoudbilal I have found the origin of this bug. I will fix it by tomorrow.
Thanks for reporting it!

jcarpent, 2018-11-25:
#608 has normally solved your issue. @hammoudbilal can you confirm it?

closed 2018-11-26 by jcarpent

hammoudbilal, 2018-11-26:
everything works perfectly, thanks