stack-of-tasks/pinocchio #610: Eigenpy installed but cannot be found

ShihaoWang, 2018-11-26:
Hi,

I was able to install the Eigenpy according to the instruction https://github.com/stack-of-tasks/eigenpy. However, when I run the cmake command to configure the pinocchio, I was told that 
Checking for module 'eigenpy >= 1.4.0'
--   No package 'eigenpy' found
Here is the result of showing the installed Eigenpy
: dpkg -l | grep -i eigenpy
ii  robotpkg-py27-eigenpy                                 1.5.0                                                amd64        
                                                                                                                   Eigen bindings to python
I am not sure how what causes this problem. Can you help me with that?

jcarpent, 2018-11-26:
You have to configure pkg-config path to let `cmake` find eigenpy.
In your case, as you are relying on the PPA of robotpkg to install EigenPy from binairies, you have to add the install path to the pkg-config path.
The EigenPy binairies are installed in /opt/openrobots. So the pkg-config files are installed in: `/opt/openrobots/lib/pkgconfig`.
You then need to add this path to your environment variable `PKG_CONFIG_PATH`
So, you just need to wirte in your bash profile the following lines as follows:
`export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/opt/openrobots/lib/pkgconfig`
You also need to that for the `LD_LIBRARY_PATH`
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/openrobots/lib`

All these details are reported here: https://stack-of-tasks.github.io/pinocchio/download.html

ShihaoWang, 2018-11-26:
Yes, after changing these two environment variables, the cmake .. returns a successful result. However, when I proceed to the make -j4 command, an error with the numpy occurs as follows:
/opt/openrobots/include/eigenpy/fwd.hpp:23:31: fatal error: numpy/numpyconfig.h: No such file or directory
 #include <numpy/numpyconfig.h>
                               ^
compilation terminated.
In file included from /home/shihao/pinocchio/build/include/pinocchio/algorithm/center-of-mass.hpp:168:0,
                 from /home/shihao/pinocchio/unittest/regressor.cpp:21:
/home/shihao/pinocchio/build/include/pinocchio/algorithm/center-of-mass.hxx:136:3: warning: unused parameter ‘model’ [-Wunused-parameter]
   getComFromCrba(const Model & model, Data & data)

Is this still a path variable problem or sth else?

Thanks,

jcarpent, 2018-11-26:
Is numpy installed on your computer?

jcarpent, 2018-11-26:
Can you show me the content of the file eigenpy.pc located in /opt/openrobots/lib/pkgconfig

ShihaoWang, 2018-11-26:
Yeah, numpy is installed and I have tried to set the PYTHONPATH to the python package path in my path (/usr/local/lib/python2.7/dist-packages/). Sure, the content in the eigenpy.pc is as follows:
prefix=/opt/openrobots
exec_prefix=/opt/openrobots
libdir=/opt/openrobots/lib
bindir=/opt/openrobots/bin
pkglibdir=/opt/openrobots/lib/eigenpy
includedir=/opt/openrobots/include
datarootdir=/opt/openrobots/share
pkgdatarootdir=/opt/openrobots/share
docdir=
doxygendocdir=

Name: eigenpy
Description: Wrapping Eigen3 -- numpy
URL: http://github.com/stack-of-tasks/eigenpy
Version: 1.5.0
Requires: eigen3 >= 3.0.5
Conflicts: 
Libs: -Wl,-rpath,${libdir} -L${libdir}  -leigenpy -Wl,-rpath,/usr/lib/x86_64-linux-gnu -L/usr/lib/x86_64-linux-gnu -lboost_python-py27
Libs.private: 
Cflags:  -I${includedir} -I/usr/include -isystem/usr/include/python2.7 -isystem/usr/lib/python2.7/dist-packages/numpy/core/include
[image]

ShihaoWang, 2018-11-26:
Okay, I think that I got it! The path leading to the core/include in this eigenpy.pc file is not the same as the one I have. Let me change it to the /usr/local/lib/python2.7/dist-packages/numpy/core/include and see what will happen.

jcarpent, 2018-11-26:
Is /usr/lib/python2.7/dist-packages/numpy/core/include pointing to an existing path? Otherwise, you have ton install numpy for all the users with pip or you can install by hand eigenpy. It is simple as installing Pinocchio from source

jcarpent, 2018-11-26:
I just saw your answer. This seems that this will solve your issue. Let’s me know!

ShihaoWang, 2018-11-26:
Yes, after I manually change the include path in the eigenpy from /usr/lib/python2.7/dist-packages/ path to /usr/local/lib/python2.7/dist-packages/, the make -j4 and make install commands successfully build this package.
Thanks a lot for your help.

jcarpent, 2018-11-26:
It’s my pleasure.

closed 2018-11-26 by jcarpent