stack-of-tasks/pinocchio #624: Forward dynamics: J and gamma in the API ... why?

nmansard, 2018-12-28:
The forward dynamics algo computes the model acceleration given some torques and contact info. It is indeed requesting the explicit jacobian J and contact drift gamma to be provided by the user, while these info can be computed from the state q,vq, also given in input.
Shouldn't we provide a simpler API where J and gamma are already computed. At least, we could provide that for 6D contact, maybe also for 3D contacts. The API might be:
forwardDynamics(model,data,q,vq,tauq,[contact_1 ... contact_N]) with [...] a list of contact, each of which being a structure with the frame id and the contact dimension:
struct ContactInfo
{ 
frame_id idx;
enum { CONTACT_3D, CONTACT 3D } contact_type; 
};
What do you think?

olivier-stasse, 2018-12-28:
Sounds good to me.

jcarpent, 2018-12-28:
I have already implemented such a solution. 
I will provide it in two weeks if you are able to wait until then.

jcarpent, 2018-12-29:
@nmansard Is it fine for you?

jcarpent, 2018-12-30:
Another answer to this issue comes from historical reasons.
The idea behind J and gamma is to enable the user to define his/her own contact model like rolling contact.
The current API is then much more flexible than the one assuming rigid contact model.

nmansard, 2019-01-11:
I agree with the historical reason, that is still valid: we should offer the largest API to the user, which is the one defined now. But I would like to see the two classical cases (rigid 3D and 6D) implemented with a simpler API.

nmansard, 2019-01-11:
I suggest to close the issue. Please re-open it if you would like to see this functionality implemented soon.

closed 2019-01-11 by nmansard