stack-of-tasks/pinocchio #640: Python 3?

jbmouret, 2019-01-16:
Hello

Does the library works with Python3? I see in the brew script that it explicitly depends on python2, but all the "modern" frameworks in python (e.g. pytorch) require python3.

Thanks.

jcarpent, 2019-01-16:
Yes, Pinocchio is compatible with Python 3. I need to update the https://github.com/Gepetto/homebrew-gepetto/blob/master/Formula/pinocchio.rb formula to take into account the latest Pinocchio version as well as Python 3.

@jbmouret Thanks for this kind report.

nim65s, 2019-01-16:
Hi @jbmouret,

We have been working on assuring compatibility with Python3, but sadly it is still not a first class citizen.

To develop a bit @jcarpent answer, I would say that Pinocchio should be working with Python 3, but:
- It is not as tested as Python 2 because our user base is still using almost only Python 2 ;
- [An annoying bug in eigenpy](https://github.com/stack-of-tasks/eigenpy/issues/35) makes the interpreter segfault when you are done with it ;
- Boost-Python bindings are compiling, but on some distributions cmake might not find the right boost-python library version (eg. https://github.com/jrl-umi3218/jrl-cmakemodules/pull/172 has been merged after the v2.0.0 Release)
- Most scripts should be working, but I recently found a few python2-only syntax in some part of them in the v2.0.0 Release, and I will provide a PR soon
- The viewer should be working, but only in the latest git versions, we still need to make a new release including [the last PR for Python 3 compatibility](https://github.com/Gepetto/gepetto-viewer-corba/pull/92).

jbmouret, 2019-01-16:
Thanks. So far I managed to compile half of the source with: `cmake .. -DPYTHON_EXECUTABLE=/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/bin/python3.7 -DPYTHON_LIBRARY=/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib/python3.7/config-3.7m-darwin/libpython3.7m.dylib -DPYTHON_INCLUDE_DIR=/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/include -DCXXFLAGS=-I/usr/local/Cellar/numpy/1.15.4/lib/python3.7/site-packages/numpy/core/include/`

But apparently it does not find a include of numpy (that I have):
`/usr/local/include/eigenpy/fwd.hpp:23:10: fatal error: 'numpy/numpyconfig.h' file not found`

I will try a bit more...

jcarpent, 2019-01-16:
It is because you've installed eigenpy for Python2 (through brew I guess).
I've just updated the homebrew-gepetto repository for this purpose.

jcarpent, 2019-01-16:
@jbmouret I've just updated the Formulas for pinocchio.
You can now install pinocchio with Python3 support through:
`brew install pinocchio-python3`

jbmouret, 2019-01-16:
Perfect! it's working.

closed 2019-01-16 by jbmouret