stack-of-tasks/pinocchio #641: More constructors for Data object?

gabrielebndn, 2019-01-16:
Currently, class `Data` only has one constructor, taking a `Model` object as input.
It may be useful to add some other ways of building the object. For instance:
- A default constructor
- A copy constructor
- An assignment operator
- A new method called something like `setModel(Model)`, which would basically make the `Data` object identical to a newly-created object

This would allow more flexibility.
In particular, the lack of a default constructor means it is not possible to create a data object whose model is only determined later (for instance, by reading a file). This is a problem, for instance, when the `Data` object is a member object in another class: either the model is known at instance creation, or one is forced to use a pointer and create the `Data` object with the operator `new`

gabrielebndn, 2019-01-16:
Is there a specific reason why this was not done? Otherwise I can do it

jcarpent, 2019-01-16:
Copy and assignment operators are already implicit.
The copy constructor is also implicit.

The default constructor would mean nothing because a data is defined from a model. This is how Pinocchio has been thought initially. In addition, if you provide later the information of the model, then it means that you will do dynamic memory allocation, which may be bad in some sense. That's also why we have decided to not put such a constructor. 

The setModel can be dangerous too. Because, it implies that you can online change the model, meaning that you will change all the content of the vectors, mainly the data.joints and so on.

So the idea is really to assign a unique data to each model, and that a data cannot change of owner in some sense.

gabrielebndn, 2019-01-16:
I just was not sure that the implicit copy/assignment operators would work fine, but on a second thought they should.

I was thinking of a particular use-case where those features might be needed, but I think I solved my problem.

Thanks, I'll close the issue

closed 2019-01-16 by gabrielebndn