stack-of-tasks/pinocchio #648: Derivatives of impulse dynamics

nmansard, 2019-01-21:
I need to compute the derivatives of the impulse dynamics, i.e of the function:
i(q,dv) = M(q) dv + J(q)^T f 
I need the derivative of i with respect to q di/dq. 
i(q,dv) can indeed be computed from rnea: i(q,dv) = rnea(q,v=0,a=dv) when gravity is null, or less efficiently i(q,dv) = rnea(q,v=0,a=dv) - rnea(q,v=0,a=0). I can of course either turn gravity of just before calling drnea, or calling drnea twice with a=dv and a=0. I wonder if somebody would have a better idea?

nmansard, 2019-01-27:
Ok. I can wait but would need this feature soon. 
I can surely implement what I described in the top text, from drnea, by removing unecessary terms related to gravity and centrifugal forces, but would certainly prefer that you provide the full impact derivatives if you already have them.

jcarpent, 2019-01-27:
I let it open because it is a requested feature.

jcarpent, 2019-01-27:
By the way, you can just do better than you suggest for getting the impulse dynamics. I will provide it soon.

proyan, 2019-08-27:
Hi Justin,
Is there any activity on this issue?

jcarpent, 2022-05-04:
Done in pinocchio-3x which should be released soon.

closed 2022-05-04 by jcarpent