stack-of-tasks/pinocchio #651: Distance / projection on configuration manifold

nmansard, 2019-01-27:
I need to put a constraint in an optimization problem to keep my configuration parameter on the configuration manifold. Typically force the quaternion to be unit. In Python.
I can off course implement that explicitly (if the joint is a freeflyer, then constraint the norm of q[3:7] to be 1) but I am wondering if such a "distance-to-manifold" function is already implemented in the C++ api. Surely, HPP-constraint is imposing something like that when projecting. 
- could we have a pinocchio.distanceToManifold(rmodel,q) implemented in the joint API? The function would return 0 if q is on the manifold (eg the quaternion is unit) and a strictly positive value otherwise.
- should we provide a projectToManifold(rmodel,q) algorithm implemented? The function would return q if q is on the manifold (e.g. the quaternion is unit) or a configuration qproj on the manifold which is somehow nearby the input q.

jcarpent, 2019-01-27:
As far as I know, there is no reason to do that. You should instead work on the tangent manifold directly and use the dedicated integrators: that all the goals of Pinocchio, i.e. working on the correct geometry!

If you need to normalize your configuration vector, then there is in the Lie group API something called `normallize`, but this is NOT inside the Joint API and it is not recommended to use inside non-linear solvers.
The distance to manifolds is then easy to get from this function.

jcarpent, 2019-01-29:
I close this issue. Feel free to reopen it.

nmansard, 2019-01-30:
I dont agree. Most solvers do not provide a manifold-able interface. Here I was using SciPy SLSQP, which is a good solver, but must work with the constraint. Alternatively, you can work in the tangent space to a reference point that will not be automatically adapted during the optimization, and this is costly and can lead to discrepancy when working to far from the reference point.

@jmirabel  How are you doing in HPP? You have a dedicated manifold-able solver?

jmirabel, 2019-01-30:
>  How are you doing in HPP? You have a dedicated manifold-able solver?

Yes.

jcarpent, 2019-02-13:
@nmansard Do you want to add additional functions to perform this projection?
From my side, the function `normalize`does the trick for us and it is not needed to add the distance version, as it is just a call to `normalize` and `distance` algos.

closed 2019-02-19 by jcarpent