stack-of-tasks/pinocchio #654: urdfdom shared pointer redefinition on Ubuntu 18.04

wxmerkt, 2019-01-30:
On Ubuntu 18.04, the ROS-shipped urdfdom already has the types (casting) that are defined in `types.hpp` when using the define to turn std_shared_ptr on. Pinocchio can be compiled and used when commenting out (the same definitions are shipped with urdfdom):

```
  template<class T, class U>
  PINOCCHIO_URDF_SHARED_PTR(T) const_pointer_cast(PINOCCHIO_URDF_SHARED_PTR(U) const & r)
  {
#ifdef PINOCCHIO_URDFDOM_USE_STD_SHARED_PTR
    return std::const_pointer_cast<T>(r);
#else
    return boost::const_pointer_cast<T>(r);
#endif
  }
  
  template<class T, class U>
  PINOCCHIO_URDF_SHARED_PTR(T) dynamic_pointer_cast(PINOCCHIO_URDF_SHARED_PTR(U) const & r)
  {
#ifdef PINOCCHIO_URDFDOM_USE_STD_SHARED_PTR
    return std::dynamic_pointer_cast<T>(r);
#else
    return boost::dynamic_pointer_cast<T>(r);
#endif
  }
  
  template<class T, class U>
  PINOCCHIO_URDF_SHARED_PTR(T) static_pointer_cast(PINOCCHIO_URDF_SHARED_PTR(U) const & r)
  {
#ifdef PINOCCHIO_URDFDOM_USE_STD_SHARED_PTR
    return std:static_pointer_cast<T>(r);
#else
    return boost::static_pointer_cast<T>(r);
#endif
  }
```

jcarpent, 2019-01-30:
This is shipped with urdfdom since 0.4.0
I guess you are not able to detect the version of urdfdom with pkg-confit. Can you show the value of URDFDOM_VERSION in cmake?

wxmerkt, 2019-01-31:
The compilation of Pinocchio is actually fine, packages that depend upon it using pkg-config suffer - the issue appears to be that catkin generates its own pkg-config files that do not contain the definitions added via `PKG_CONFIG_APPEND_CFLAGS`. 

I have implemented a `CFG_EXTRAS` [CMake workaround](https://github.com/ipab-slmc/pinocchio_catkin/commit/9f2aaa3fcfab9b673502ac2bdebfc69e45a0670e) that mimics `PKG_CONFIG_APPEND_CFLAGS` logic for URDFDOM (no Lua or CppAD support via catkin yet).

closed 2019-01-31 by wxmerkt

jcarpent, 2019-01-31:
Ok. Is there something special in their pkg file? Would it be possible to overwrite it?

wxmerkt, 2019-01-31:
The challenge is that we are using an `ExternalProject_Add` logic for packaging and then copy/install the required files. The Pinocchio-generated pkg-config have all paths defined at the top of the pkg-config, and we would need to replace those when overwriting the pkg-config file.

jcarpent, 2019-01-31:
Certainly, we have to adapt the jrl-cmake-modules to be more compliant with Catkin.

olivier-stasse, 2019-01-31:
Dear Wolfgang I am not sure I get the problem.
For now what I have seen is the following problem. When installing through binaries on bionic, pinocchio is also installing robotpkg-urdfdom which is the 0.3.0 version. This is problem on our side that we can probably fix and use the default bionic version.

wxmerkt, 2019-01-31:
Hi Olivier,
We are using a different approach and not `robotpkg`. The aim is to be able to fully build `pinocchio` in a catkin workspace and depend on it by `find_package(catkin REQUIRED COMPONENTS pinocchio)` (in lieu of `pkg_check_modules(pinocchio)`). The background is that all required dependencies can be resolved via `rosdep install -iy --from-path` - and most users right now suffered from resolving all dependencies in a trial-and-error fashion (plus requirement to install into a path and then manually add it to your `PKG_CONFIG_PATH`, `LD_LIBRARY_PATH`, etc.). Our mid-term goal is to package `pinocchio` as a catkin package.

The current issue is that `catkin` does not correctly export the compile definitions into the `pkg-config` files it creates. As such, packages that depend on the catkinised pinocchio via `pkg-config` do not get all compile definitions (for now). For those depending on it via catkin are fine. As thus, I am still working on fixing the pkg-config definitions.

olivier-stasse, 2019-01-31:
If this is just a matter or adding a package.xml file in the main repository following
http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty
then we could probably consider it.

wxmerkt, 2019-01-31:
We already completed the packaging:

- https://github.com/ipab-slmc/eigenpy_catkin
- https://github.com/ipab-slmc/pinocchio_catkin

We are happy to maintain this and release it upstream via the ROS buildfarm. For now the only thing not working is _the catkin-generated pkg-config as it does not support the add_definitions added preprocessor directives_. This is due to a limitation with the `.pc` files generated by catkin itself. Using CMake already works fine.

olivier-stasse, 2019-01-31:
Excellent !
Thanks a lot for this work.
Just out of curiosity, how are you handling hpp-fcl ? I was a bit worry because this is a fork from fcl. It has a different namespace so this should not a problem. I guess that in your current setup this not selected.

wxmerkt, 2019-01-31:
We are not using `hpp-fcl` for the packaging or in the catkin wrapper. It could be as easily exposed though (we are exposing the latest `fcl` as `fcl_catkin`, which is also available via ROS binaries). What are the differences to the mainstream fcl and are there any plans to upstream them?

jcarpent, 2019-01-31:
I think @jmirabel and @florent-lamiraux are the right persons to answer this last question?

jmirabel, 2019-01-31:
> What are the differences to the mainstream fcl 

Additional features:
- collision with tolerance (i.e. in collision if d < thr, and not d < 0, where thr >= 0),
- computation of a lower bound of the distance in collision queries.

Other changes:
- removal of dead code or code not relevant to collision detection library,
- restriction to only one GJK algorithm (remove dependency to libccd),
- bug fixes + unit-test in collision / distance query between pairs of primitive we use (Box, Cylinder, Capsule, Triangle and OcTrees).
- homogenization of the result of collision / distance queries.

> are there any plans to upstream them?

Merging is a good idea but it requires a lot of work. We do not have the resources to do that.
The fork comes from the unresponsiveness of FCL maintainers a few years ago. When it became active again, we were already far ahead.
Our current roadmap is directed towards a smaller and cleaner library.

wxmerkt, 2019-01-31:
Thank you very much for the detailed summary - and those improvements are much needed and appreciated (alas, I only learnt about the advantages now - we spent a lot of time looking into debugging FCL issues).

jcarpent, 2019-01-31:
Then, @wxmerkt do you think possible to put `hpp-fcl`inside ROS in order to provide Pinocchio with collision checking for instance?

jmirabel, 2019-01-31:
Currently, the only possible collision between FCL and hpp-fcl is the C++ namespace.

It is blocking only for people who plan to use both FCL and hpp-fcl in their code. It will be a problem if, by means of dynamic library loading, both libfcl.so and libhpp-fcl.so gets loaded. The application will crash in such case.

We should probably change the namespace in hpp-fcl.

jmirabel, 2019-01-31:
I opened an issue: https://github.com/humanoid-path-planner/hpp-fcl/issues/51

wxmerkt, 2019-01-31:
Thank you for pointing this out. I have created a wrapper for hpp-fcl and will test it with Pinocchio for packaging. It would be good though if the namespace clash is resolved before we target a release.