stack-of-tasks/pinocchio #658: Joint-configuration algorithms are performing memory allocation

gabrielebndn, 2019-02-04:
A lot of emphasis is put on the idea that Pinocchio algorithms *never* perform any memory allocation, relying on storage inside `Data` or in the user passing the required memory,  as in the case of `getJointJacobian(model, data, jointId, rf, J)`, where the result is written in matrix `J`.

However, pretty much all algorithms contained in `joint-configuration.hpp` do not follow this pattern. Instead, the result vector is allocated and returned.

Should we adopt a pattern which mimics the Jacobian case? Do you think it is worth it? In this case, should we deprecate the current signatures, or leave them as a more friendly option? This last question arises from the fact that in most cases `randomConfiguration` and `neutral` are just used as one-shot calls at the beginning of a program, and forcing the user to pre-allocate their memory would just make their use clunkier with no real benefit.

jcarpent, 2019-02-04:
For simplicity, both should exist, an extended signature with output results in the arguments, and the simpler one, with the default return value. What do you think?

gabrielebndn, 2019-02-04:
I agree

jcarpent, 2019-02-13:
@gabrielebndn Can you take in charge the corresponding PR?

gabrielebndn, 2019-02-13:
Ok. Will do it as soon as possible

jcarpent, 2019-02-15:
solved by #679

PR #679 (2019-02-15): Reimplement joint-configuration algorithms, fix #658
Fix #658.
Now all joint-configuration algorithms have two versions: one with return value as argument, and one which allocates memory.

Notice that most algorithms will probably give wrong results if the output is the same object as one of the inputs. This is true in particular for `integrate`. I might implement `integrateInPlace` in the future.

Additionally, I re-implemented `distance` so that it does not employ the auxiliary vector provided by `squaredDistance`.
Also, I am providing the square of the distance with method `squaredDistanceS [... cut]
commit bf9e46c3 (2019-02-15): Merge pull request #679 from gabrielebndn/topic/jointconfig

Reimplement joint-configuration algorithms, fix #658
closed 2019-02-15 by jcarpent