stack-of-tasks/pinocchio #660: Compiling pinocchio v2.0.0 and RobotWrapper

cmastalli, 2019-02-05:
RobotWrapper doesn't work with default cmake settings for pinocchio v2.0.0. compilation. Indeed, pinocchio (Python) module haven't functions such as `buildGeomFromUrdf`. 

I figured out that it's needed to enable BUILD_UTILS in the cmake setup. I wonder if there is any reason for this, or if this is a compilation bug.

jcarpent, 2019-02-05:
Do you have installed hpp-fcl?

cmastalli, 2019-02-05:
No, I don't

jcarpent, 2019-02-05:
`buildGeomFromUrdf` is only compiled if hpp-fcl is found. Because it loads meshes and so on. But, the `RobotWrapper` should still work.

jcarpent, 2019-02-05:
Can you provide a self-content example?

jcarpent, 2019-02-05:
@cmastalli do you still face this issue?

closed 2019-02-06 by jcarpent

cmastalli, 2019-02-06:
I just want to report that I don't face this issue if I install hpp-fcl.