stack-of-tasks/pinocchio #667: ssh-path in .gitmodules limits easy access for certain users.

markusgft, 2019-02-13:
Hi all,
when cloning the source-code of the current master, certain users may encounter the following problem: currently, there is a SSH-address hard-coded in ".gitmodules", namely
`
url = git://github.com/jrl-umi3218/jrl-cmakemodules.git
`
From within some institutions and universities, it is impossible to access external sources via SSH. It would be great if you could change the URL to
`url = https://github.com/jrl-umi3218/jrl-cmakemodules`, which avoids unnecessary overhead when cloning pinocchio.

Keep up the great work! :)

jcarpent, 2019-02-13:
Ok. Sounds reasonable.
I will fix it now.

jcarpent, 2019-02-13:
Solved by #669.

nim65s, 2019-02-13:
For me, the solution is `url = ../../jrl-umi3218/jrl-cmakemodules`
This way, if you use `git clone --recursive https://github.com/stack-of-tasks/pinocchio` the submodule will be cloned in https, and if you use `git clone --recursive git@github.com:stack-of-tasks/pinocchio` the submodule will be cloned in ssh.

jcarpent, 2019-02-13:
Looks fine to me too.
I will update my PR with it. @nim65s Thanks for this nice detail.

jcarpent, 2019-02-13:
Solved by #670.

closed 2019-02-13 by jcarpent

markusgft, 2019-02-15:
Thanks for addressing this issue. Would you mind, however, to also cherry-pick this to the master branch?

jcarpent, 2019-02-20:
@markusgft Sorry, I did not see your comment. I will do that tomorrow.

jcarpent, 2019-02-21:
@markusgft Done!