stack-of-tasks/pinocchio #668: Passing static constructors of Eigen-types to pinocchio methods

markusgft, 2019-02-13:
Hi all,
this is not an urgent issue, but for the sake of documentation, let me note it here:

For newer Eigen versions (tested with 3.3.5 and 3.3.3), there is a compilation error when passing statically constructed arguments to pinocchio methods. 

For example, if you try to compile
`pinocchio::forwardKinematics(model_, data_, JointPosition_t::Zero());`

you will obtain the compiler-error
`
/usr/local/include/pinocchio/multibody/joint/joint-free-flyer.hpp:244:95: error: ‘Options’ is not a member of ‘Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, 7, 1, 0, 7, 1> >’
       typedef typename Eigen::Quaternion<typename ConfigVector::Scalar,ConfigVector::Options> Quaternion;
                                                                                               ^
/usr/local/include/pinocchio/multibody/joint/joint-free-flyer.hpp:245:44: error: ‘Options’ is not a member of ‘Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, 7, 1, 0, 7, 1> >’
       typedef Eigen::Map<const Quaternion> ConstQuaternionMap;
`

If you instead use the argument
`const JointPosition_t zeroPos = JointPosition_t::Zero();
pinocchio::forwardKinematics(model_, data_, zeroPos);`
compilation works fine.

jcarpent, 2019-02-13:
Ok. I won't have time to fix today, but I will do it in the next few days.
@markusgft Thanks a lot for raising this important issue.

jcarpent, 2019-02-20:
solved by #689.

PR #689 (2019-02-20): joints: fix calc with static constructor
fixes #668
closed 2019-02-20 by jcarpent