stack-of-tasks/pinocchio #674: RNEA test fails on 18.04

nim65s, 2019-02-13:
Hi,

If I add two `std::cout` to the rnea test on 18.04, I get those results:
```
test 5
    Start  5: rnea

5: Test command: /root/pinocchio/build/unittest/rnea
5: Test timeout computed to be: 9.99988e+06
5: Running 5 test cases...
5: (the time score in debug mode is not relevant)  1366 us
5: data_rnea_fext.tau:  142.506  57.5557  225.933  53.1439   7.1482 -114.178  46.5939  4.61461 -43.4066  44.4801  3.36533  9.61271  5.50944  21.5901  18.3586  4.48687   8.8325  -13.126  105.974 -10.4501 -19.7189   -18.25  17.4811  17.3303 -1.77389 -2.85662 -64.3231  32.0504 -4.73482 -13.2111  11.8662 -5.55383
5:
5: tau_ref:   140.96  57.6699   225.37  53.1661  5.70757 -115.183   46.722  4.65187 -43.5309   43.506  4.68069  9.61271  6.20521  20.7955  19.0152  4.03153  9.15568  -13.126  105.974 -10.4501 -19.7189   -18.25  17.4811  17.3303 -1.77389 -2.85662 -64.3231  32.0504 -4.73482 -13.2111  11.8662 -5.55383
5:
5: /root/pinocchio/unittest/rnea.cpp(162): error: in "rneaTest/test_rnea_with_fext": check tau_ref.isApprox(data_rnea_fext.tau) has failed
5:
5: *** 1 failure is detected in the test module "rneaTest"
```

jcarpent, 2019-02-13:
Which commit version?

nim65s, 2019-02-13:
It was on https://github.com/nim65s/pinocchio/commit/e5a390e75cfe56c02683092787e479e459f65c4b, which is #645 + #673 ; but I can run this test on other commits if you prefer

jcarpent, 2019-02-13:
Yes, sure.

nim65s, 2019-02-18:
I'm able to reproduce this on the current devel branch:
```
5:   140.96  57.6699   225.37  53.1661  5.70757 -115.183   46.722  4.65187 -43.5309   43.506  4.68069  9.61271  6.20521  20.7955  19.0152  4.03153  9.15568  -13.126  105.974 -10.4501 -19.7189   -18.25  17.4811  17.3303 -1.77389 -2.85662 -64.3231  32.0504 -4.73482 -13.2111  11.8662 -5.55383
5:  142.506  57.5557  225.933  53.1439   7.1482 -114.178  46.5939  4.61461 -43.4066  44.4801  3.36533  9.61271  5.50944  21.5901  18.3586  4.48687   8.8325  -13.126  105.974 -10.4501 -19.7189   -18.25  17.4811  17.3303 -1.77389 -2.85662 -64.3231  32.0504 -4.73482 -13.2111  11.8662 -5.55383
```

I created a [Dockerfile](https://github.com/Gepetto/gepetto-utils/blob/master/docker/pinocchio/Dockerfile) to help reproduce this.
To run tests on 18.04 / python 2, you can just use `docker run --rm -it gepetto/utils:pinocchio-18.04`; and if you want to enter into a container to debug it: `docker run --rm -it gepetto/utils:pinocchio-18.04 bash`

Available images are: 
- `pinocchio-14.04`
- `pinocchio-16.04`
- `pinocchio-18.04`
- `pinocchio-18.04-py3`

nim65s, 2019-02-18:
(launching a `docker run` on an image which is not available locally will automatically download it)

jcarpent, 2019-02-18:
Perfect. I will have a look today. Thanks a lot for this help.

nim65s, 2019-02-18:
I forgot to add that now, the test `joint-configurations` is also failing

jcarpent, 2019-02-18:
Thanks for the report. I will have a look at it too.

jcarpent, 2019-02-21:
Solved by #690.

closed 2019-02-21 by jcarpent