stack-of-tasks/pinocchio #677: Python unit tests failing in non-Release mode

gabrielebndn, 2019-02-14:
Python unit tests [bindings_com.py](https://github.com/stack-of-tasks/pinocchio/blob/devel/unittest/python/bindings_com.py), [bindings_dynamics.py](https://github.com/stack-of-tasks/pinocchio/blob/devel/unittest/python/bindings_dynamics.py) and [explog.py](https://github.com/stack-of-tasks/pinocchio/blob/devel/unittest/python/explog.py) seem to pass in Release mode, but they fail badly in other compilation modes.

In particular:  when assertions are enabled (not only in Debug mode, tested on Mac OS 10.11), `bindings_com.py` and `bindings_dynamics.py` raise an assertion.
In Debug mode, additionally, `explog.py` segfaults (tested on Ubuntu 16.04).

Although the tests nominally work in Release mode, these error might reflect some subtler problems which were somehow not appearing in the unit tests, therefore it is worth to investigate this.

I will take a look

jcarpent, 2019-02-14:
This also means that we need to enforce the debugging check with Travis too.

gabrielebndn, 2019-02-14:
I do agree

jmirabel, 2019-02-14:
To avoid making Travis take longer by increasing the number of compilations, you also have the possibility to force the compilation flags.

Something like `-O3`, without `-DNDEBUG` and `-g`, would keep the assertions while still doing some optimization. It should do the work.

gabrielebndn, 2019-02-14:
I think @jmirabel's solution is a good compromise. At least the assertions will be checked.

jcarpent, 2019-02-14:
I would add that using O3 and no hard compilation flags won't provide the computation tree, and they may lead to different results.
I think it would be better to add also the Debug flag in Travis too, at least on 16.04. By the way, we may think to stop 14.04, as it will be deprecated in two months now.

gabrielebndn, 2019-02-14:
Solved by #678

PR #678 (2019-02-14): Fix python unit tests, fix #677
Fixes #677. Related to #673. The error on `explog.py` was actually due to the binding of `SE3.isIdentity()` which was crashing. I do not understand why there was this problem and why only in Debug mode, but I fixed it.

The error on `bindings_com.py` and `bindings_dynamics.py` was just a trivial error in the unit test itself (wrong vector size). I corrected it. Which makes me think, wouldn't it be better to enable assertions the Python library? If yes, is there any way to treat a failing assertion as an exception instead of having Python crash completely when on [... cut]
commit c412138f (2019-02-14): Merge pull request #678 from gabrielebndn/topic/fixpythontest

Fix python unit tests, fix #677
closed 2019-02-14 by gabrielebndn