stack-of-tasks/pinocchio #686: v2.0 on robotpkg

markusgft, 2019-02-19:
Dear pinocchio developers,
I have adopted pinocchio v2.0 and find it an outstanding and well-made tool.
Is there any update on when you are planning to release v2.0 via robotpkg?
Thanks!

jcarpent, 2019-02-19:
hi @markusgft. Thanks for this question. The main person in charge of this release on robotpkg is @nim65s. 
You may also think to use the release on ROS. In this case, @wxmerkt is the right person to help you.

wxmerkt, 2019-02-19:
v2.0 has been released for ROS Kinetic and Melodic. `eigenpy` is already in the default repositories and `pinocchio` is currently in the shadow repository and will be after the next sync (for Kinetic, early next week, Melodic likely shortly thereafter).

nim65s, 2019-02-19:
Hi @markusgft,

Thanks for your views on Pinocchio !

About the release on robotpkg, we are waiting for all of the packages of the Gepetto group  to be compatible with it (we don't want to release anything on robotpkg that might break our robots…). This imply making a release of all of them : http://homepages.laas.fr/gsaurel/graph.svg
Obviously, this takes a lot of time, but we are almost done, and I could answer your question with a "in a week", but I think I should follow Debian guidelines and say "when it's ready". Sorry about that.

All of those packages seems ready now, but we have issues with ubuntu 18.04 (cf. #674), so we will likely be in favour of a new release of Pinocchio (2.1.0 ?) with hpp-fcl 1.0 which would still not officially be compatible with ubuntu 18.04, as we have been waiting for months, and we are longing for its new functionalities.

markusgft, 2019-02-19:
Thanks all for your comments! Looking forward to the release, I will closely follow your developments. Keep up the great work! 👍

jcarpent, 2019-02-21:
@nim65s I think we can proceed now to release hpp-fcl. I will then rebase topic/fcl-release and merge it on devel.

When do you have time to do that?

nim65s, 2019-02-21:
hpp-fcl v1.0.1 source package has been released. It has not been pushed to robotpkg yet as it would break other packages that have not been released.

So I think we should merge topic/fcl-release on devel, and then make a source release of pinocchio, and then make source release of all other packages of the SoT / HPP / RBPRM, and then make a binary release of all those source releases at once, and then merge all devel branches their master.

I found a way to make the CI (with gitlab) work with the source releases when we need it, so it could help to make the pinocchio release. It should be finished tonight. If this work, everything could be really smooth.

nim65s, 2019-02-23:
ref #697

littleggghost, 2019-02-26:
Hi, @nim65s 
I wonder if the v2.0.0 can be released today? It has been a week since your first reply about this.
Or where could i read the v1.3.3 documentation like the [current docs]?(http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/index.html)

nim65s, 2019-02-26:
Hi @littleggghost 

I was waiting for @jcarpent to make a new source release, and then release this one on robotpkg.
@jcarpent : but maybe you prefer that I do it ? Or are you waiting for something ?

jcarpent, 2019-02-26:
You should first release 2.0.0 if it not online. There is no reason to have waited so long for this one.

jcarpent, 2019-02-26:
For the next release, I guess I would like to be sure that all the tests we are making are working well. Can you setup on the LAAS gitlab tests with the latest release of hpp-fcl?

nim65s, 2019-02-26:
We would have to make one release for hpp-fcl v0.7 / pinocchio v2.0 and another for hpp-fcl v1.0 v2.1. I know that it would take twice the time, and I don't see how this could be better for anybody.

The LAAS gitlab tests are working with the latest release of hpp-fcl, this is the whole point of #697

jcarpent, 2019-02-26:
Currently, a lot of people are waiting for the release 2.0.0. And we have not tested at all the release 2.1.0. So, for safety reasons, I suggest to release hpp-fcl 0.7.0 and pinocchio 2.0.0 and then we proceed to the other releases when we have time. I think the first set of release would have performed a while ago, and we cannot make people wait longer.

jcarpent, 2019-02-26:
@nim65s Concerning hpp-fcl, I just fix a bug related to Eigen version. See humanoid-path-planner/hpp-fcl/pull/61 for further details. Can you add this patch on top of robotpkg? It will then allowed the compilations under 12.04.

nim65s, 2019-02-26:
Then we have been waiting for weeks for nothing. But let's do this.

About hpp-fcl, let's continue the discussions on the PR, but yes I can add a patch.

nim65s, 2019-02-26:
Before I start… @jmirabel is it OK for you if I make a release of HPP with hpp-fcl 0.7 / pinocchio 2.0, just after that another release of HPP with hpp-fcl 1.0 / pinocchio 2.1 ?

nim65s, 2019-02-26:
For me this really looks like a waste of time. But as @jcarpent said we might have pinocchio v2 a bit earlier this way.

jmirabel, 2019-02-26:
> Before I start… @jmirabel is it OK for you if I make a release of HPP with hpp-fcl 0.7 / pinocchio 2.0

Put aside the waste of time, is this even possible ?

nim65s, 2019-02-26:
This is what I was looking at. And this looks really tricky, as all devel branches of HPP are already on hpp-fcl 1.0+ / pinocchio 2.0+

nim65s, 2019-02-26:
So I suggest to stick to the initial plan, and release pinocchio 2.1 :P

jcarpent, 2019-02-26:
I'm now running the final test [here](https://gepgitlab.laas.fr/stack-of-tasks/pinocchio/pipelines/3051). Let see the final status of CI and I will release Pinocchio then.

littleggghost, 2019-02-27:
So, today will release out?

jcarpent, 2019-02-27:
Yes, sure!

nim65s, 2019-02-27:
> So, today will release out?

Well, today @jcarpent should be able to make the source release of 2.1.0. 

Then I will be able to start the work to put this source release on robotpkg, which will lead to having a binary release. But this work is not instantaneous, as we have to test and release [a lot of other softwares that depend on pinocchio](http://homepages.laas.fr/gsaurel/graph.svg).

Most of the work is already done, so this should not take long, but don't expect to have a binary available today.

jcarpent, 2019-02-27:
The release has been pushed online. It is currently available on branch [v2.1.0](https://github.com/stack-of-tasks/pinocchio/tree/v2.1.0).
I will update master and devel branches as soon as hpp-fcl v1.0.0 is packaged.

nim65s, 2019-02-27:
I will put hpp-fcl v1.0.1 & pinocchio v2.1.0 on robotpkg ASAP.

jcarpent, 2019-03-05:
Pinocchio 2.1.0 is now available on robotpkg.

closed 2019-03-05 by jcarpent

wxmerkt, 2019-03-05:
v2.0.0 is available via ROS (without HPP-FCL) as of today as well via `sudo apt-get install ros-[kinetic/melodic]-pinocchio`

jcarpent, 2019-03-05:
Maybe, @wxmerkt you meant v2.1.0?

wxmerkt, 2019-03-05:
No, I released v2.0.0 and did not yet re-release v2.1.0. This will happen later this week (and then take 2-2.5 weeks to propagate through buildfarm).

jcarpent, 2019-03-05:
Ok. Got it. I didn't know that it will take so much time to propagate. Thanks a lot for this feedback.