stack-of-tasks/pinocchio #694: Eigen minimum version

jmirabel, 2019-02-21:
Currently, the minimum version is 3.0.4 and it is not correct. I know that 3.2.4 does not work and 3.2.92 does. Is it fine to update the minimum version to 3.2.92 (ubuntu 16.04) ?

nim65s, 2019-02-21:
I think Pinocchio is still working on 14.04 with eigen 3.2.0

jmirabel, 2019-02-21:
@ksyy had this error with 3.2.4:
```
/usr/localDev/HPP/install/include/pinocchio/multibody/joint/joint-revolute.hpp:157:63:
error: could not convert ‘Eigen::DenseBase<Derived>::Zero(Eigen::DenseBase<Derived>::Index)
  [with Derived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, 3, 1> >;
           Eigen::DenseBase<Derived>::ConstantReturnType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, 3, 1> > >;
            typename Eigen::internal::traits<T>::Scalar = double;
            Eigen::DenseBase<Derived>::Index = long int]()’
from ‘const ConstantReturnType
  {aka const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, 3, 1> > >}’
to ‘pinocchio::TransformRevoluteTpl<double, 0, 0>::LinearType
  {aka Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, 3, 1> >}’
     LinearType translation() const { return LinearType::Zero(3); };
```

nim65s, 2019-02-21:
https://travis-ci.org/stack-of-tasks/pinocchio/builds/496407364 : job .1 is trusty, and job .2 is xenial,

jmirabel, 2019-02-21:
Then, likely, there is one function which is not unit-tested.

jcarpent, 2019-02-21:
The minimal version is 3.0.5.

jmirabel, 2019-02-21:
@jcarpent Then, there is a bug at `multibody/joint/joint-revolute.hpp:157`

jmirabel, 2019-02-21:
It can be repeated by compiling hpp-pinocchio devel branch, with branch topic/fcl-release of Pinocchio and Eigen 3.2.4.

jcarpent, 2019-02-21:
The main question is in the end why @ksyy is working with 3.2.4? Because not LTS is furnished with such version.

nim65s, 2019-02-21:
This is the version of robotpkg-eigen3, so I suppose he got it from there, but I have no idea why this one was installed on his desktop.

jmirabel, 2019-02-21:
It was unintended. It was a forgotten installed version which is now deleted. I opened the issue for reporting. It is not blocking us.

jcarpent, 2019-02-21:
Solved by #695.

closed 2019-02-21 by jcarpent