stack-of-tasks/pinocchio #699: When can you release the examples?

littleggghost, 2019-02-23:
Hi,  @jcarpent  @nim65s 
The examples under /pinocchio/examples path you give now are empty, and I hope you can release them as soon,  especially the c++ version!
[image]

By the way, I really hope the v2.0.0 version with robotpkg can be released too.

Best regards!

nim65s, 2019-02-23:
Hi @littleggghost,

Those examples have not been written yet. We are trying to find a way to get them ASAP. Meanwhile, you can maybe have a look at the practical exercises:
- https://github.com/stack-of-tasks/pinocchio/tree/master/doc/d-practical-exercises for the source
- http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-practical-exercises_intro.html for a built version

About the second point, cf. #686

littleggghost, 2019-02-23:
Thanks for your quick reply! @nim65s 
The practical exercises are all python type, and I wonder if there have some c++ exercises?
And, I really want to confirm the release date of examples, one month, or half a year? 
Regards!

nim65s, 2019-02-23:
Hopefully between one month and half a year. @nmansard thinks we should organize some kind of hackathon to gather people to write those examples. We just need to find a good date for it.

littleggghost, 2019-02-23:
It's my great honor to participate in this young project.

closed 2019-02-23 by littleggghost