stack-of-tasks/pinocchio #700: Serialization of Model in Pinocchio

jcarpent, 2019-02-25:
I would like to raise the question of serialization of Model in Pinocchio.
A lot of users are asking for this feature, which seems to be common for many applications.

I suggest relying on Boost.Serialization to achieve this objective, which will require to also serialize Spatial classes.
Concerning Data, I do not think worthy to serialize such a class, because data contained in it are just temporary for computations.

What do you think?

jmirabel, 2019-02-25:
I implemented serialization of BVH objects in hpp-fcl, using Boost. It wouldn't be hard to extend it to shapes. It is very easy to serialize Eigen Matrices so I guess spatial classes won't be hard either.

I haven't been very far in comparing with other serialization libraries. It was really amazed by the easiness of use of Boost.Serialization.

olivier-stasse, 2019-02-25:
I would also be for boost::serialization, mostly because it would allow to have a clean separation with specific format. Maintaining serialization for various formats is known to be a pain.

jcarpent, 2019-02-25:
OK. Thanks for your feedback.
I will then rely on Boost.Serialization, and make explicit serialization for binaries, XML and text files.

proyan, 2019-02-27:
I have seen before that boost::serialization is not forward compatible. This was especially a problem when I serialized an object in 16.04 (boost 1.58) and deserialized in 14.04 (boost version 1.54). I can forsee that this would be a problem again when alternating between 16.04 and 18.04 (boost version 1.65). And I don't think the solution is in the pipeline. See for reference https://github.com/boostorg/serialization/issues/88

Please keep this is mind in case an issue is raised in the future because of this.

jcarpent, 2019-02-27:
@proyan Thanks for this useful feedback. But I guess you were using binaries and not text or XML?

proyan, 2019-02-27:
I was using text serialization.

jcarpent, 2019-03-11:
Solved by #734

closed 2019-03-11 by jcarpent