stack-of-tasks/pinocchio #701: Build v2.0.0 from source?

littleggghost, 2019-02-26:
Hi, all
Your work is very remarkable and admirable, and your accomplishment is higher, performance is better, and it is more suitable for robot development, compared to ODE, Bullet, etc.
Your physics engine is really amazing, I really can't help but use them now. Considering that v2.0.0 has not been released yet, I plan to compile it myself. But I meet some errors:
[image]
Could you help me please?

jcarpent, 2019-02-26:
You forgot to include the git submodule with your download.
You can do `git submodule init && git submodule update`

littleggghost, 2019-02-26:
Really thanks!

jmirabel, 2019-02-26:
@jcarpent Maybe we should add
```cmake
IF(NOT EXISTS cmake/base.cmake)
  MESSAGE(FATAL_ERROR "Please run the following command first:\n\ngit submodule update --init\n")
ENDIF()
```

jcarpent, 2019-02-26:
Yes, sure! I will add it! Thanks!

littleggghost, 2019-02-26:
I am sorry to reopen this issue because I meet some further problems while compiling it. 
The error says:
`/home/xxx/pinocchio/build/include/pinocchio/parsers/urdf/model.hxx:30:52: error: ‘LinkConstSharedPtr’ in namespace ‘urdf’ does not name a type
       FrameIndex getParentJointFrame(const ::urdf::LinkConstSharedPtr link,`
I have checked the **urdfdom** version which version is v0.4.1. 
I alse check files under /usr/include/urdf_model folder and I didn't find _LinkConstSharedPtr_ too.
Maybe it less some header files?

jcarpent, 2019-02-26:
Have you already installed Pinocchio on your computer?

jcarpent, 2019-02-26:
_LinkConstSharedPtr_ is declared in `urdf_model/types.h` through the macro `URDF_TYPEDEF_CLASS_POINTER`.

littleggghost, 2019-02-26:
I haven't install Pinocchio v2.0.0 successfully since I haven't built it successfully.
I'm sorry I find no _LinkConstSharedPtr_ under /usr/include/urdf_model/types.h
[image]

jcarpent, 2019-02-26:
Yes, it is done through the `URDF_TYPEDEF_CLASS_POINTER` macro. You can have a look to its definition.

jcarpent, 2019-02-26:
@littleggghost By the way, we are about to make the new release of Pinocchio, it is a matter of hours now. We just internally validate that Pinocchio is passing on 14.04, 16.04 and 18.04 and will be then able to publish the release and provide the binaries.

jcarpent, 2019-02-26:
Which feature of Pinocchio do you exactly need?
Because currently, you can deal with Pinocchio 1.3.3 on robotpkg.

littleggghost, 2019-02-26:
Yes, you are right. I find it.
My favorite points is Contact dynamic in c++ version, and the GUI for visualization.  
Well, with 1.3.3, I less the c++ examples and I meet errors before while compiling it. Some parameter of functions may have changed. I also can't find the documentation of 1.3.3.

jcarpent, 2019-02-26:
Provide a fix through #703

jcarpent, 2019-02-26:
@littleggghost We will do our best to have a clean solution in a few hours. It is indeed hard to understand why it is not compiling on your system. Have you looked at https://stack-of-tasks.github.io/pinocchio/download.html?

littleggghost, 2019-02-26:
I directly check this [link](http://robotpkg.openrobots.org/packages/debian/pub/dists/xenial/robotpkg/binary-amd64/).

littleggghost, 2019-02-26:
Hi, @jcarpent 
I forget to tell you I add `set (CMAKE_CXX_STANDARD 11)` in CMakeLists.txt, or I will meet errors about **std**.

jcarpent, 2019-02-26:
Why? There is no reason to add it. Pinocchio is compatible with C++98 and laters.
What is your compiler?

littleggghost, 2019-02-26:
My gcc and g++ version are 5.4.0.

jcarpent, 2019-02-26:
Can you try to install Pinocchio without adding `set (CMAKE_CXX_STANDARD 11)`please?

littleggghost, 2019-02-26:
The initial error I meet is `/opt/ros/kinetic/include/eigenpy/fwd.hpp:21:22: fatal error: Eigen/Core: No such file or directory`

jcarpent, 2019-02-26:
In the root of the Pinocchio, can you show the output of:
`mkdir build-debug && cd build-debug && cmake ..`

littleggghost, 2019-02-26:
The output is:
[debug.txt](https://github.com/stack-of-tasks/pinocchio/files/2905522/debug.txt)

littleggghost, 2019-02-26:
`cmake ..` works normal, but crashed in `make -j4`.

jcarpent, 2019-02-26:
It seems that you have several versions of urdfdom at least, Eigen too. You should clean your set up on your computer before going further.
The errors of compilation you have are then expected.

littleggghost, 2019-02-26:
I use a clean environment now. The output of _cmake .._ is: 
[debug.txt](https://github.com/stack-of-tasks/pinocchio/files/2906018/debug.txt)

jcarpent, 2019-02-26:
You need to install first eigenpy, as it is mentioned by the output of cmake.

littleggghost, 2019-02-27:
I have installed eigenpy through `sudo apt install robotpkg-py27-eigenpy`, but it still the same:
`-- eigenpy >= 1.4.0 is required.
-- Checking for module 'eigenpy >= 1.4.0'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:290 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:549 (ADD_DEPENDENCY)
  bindings/python/CMakeLists.txt:41 (ADD_COMPILE_DEPENDENCY)
`

littleggghost, 2019-02-27:
Ohh, I forgot to add
`export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/opt/openrobots/lib/python2.7/site-packages:$PYTHONPATH`

littleggghost, 2019-02-27:
Thanks @jcarpent from my heart! 
Thanks all your hardwork and patience!

closed 2019-02-27 by littleggghost