stack-of-tasks/pinocchio #706: Error about collision checking in practical-exercises?

littleggghost, 2019-02-28:
Hi, there
I recently do the practical-exercises as your tutorial says [4.3) Collision checking](http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-practical-exercises_4-dyn.html)
But when I run in python, I meet an error:
`  File "example_q4_contact.py", line 22, in <module>
    dist = robot.checkCollision(idx)
  File "/home/zyg/pinocchio/doc/d-practical-exercises/src/robot_hand.py", line 295, in checkCollision
    dist = va.collision(vb, self.data)
  File "/home/zyg/pinocchio/doc/d-practical-exercises/src/robot_hand.py", line 61, in collision
    if all(0 <= ab <= 1):
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
`
Wish you have a nice day!

jcarpent, 2019-02-28:
Will fix it. Thanks for this report.

jcarpent, 2019-02-28:
What do you mean by tutorial?

littleggghost, 2019-02-28:
The way used for collision checking in `robot_hand.py` is not the FCL way. Maybe use FCL in `robot_hand.py` will better as a tutorial. 
We should use native API as much as possible, rather than "make" a new one.

jcarpent, 2019-02-28:
You're right. We wanted to mask a lot of FCL API to allow end users.
But it remains possible to use the classic API of FCL.

Maybe, it would be nice for someone to do it. @littleggghost Do you think you may write such an example? Thanks in advance for your contribution to Pinocchio.

littleggghost, 2019-02-28:
Well, I will do that if I have the time. As a beginner, there is still a long way to go.

jcarpent, 2019-02-28:
If you already know FCL, it just a matter of seeing how the functions for collision detection are working. All these details are located here: [here](https://github.com/stack-of-tasks/pinocchio/blob/9f7aba293c3df696dc9896e1b8a7b88b937f2e0f/src/algorithm/geometry.hxx#L29) for updating the placement of the collision meshes and [here](https://github.com/stack-of-tasks/pinocchio/blob/9f7aba293c3df696dc9896e1b8a7b88b937f2e0f/src/algorithm/geometry.hxx#L53) for calling FCL funtions.

And of course, we can help you.

littleggghost, 2019-02-28:
Is these functions have python wrapper now?

littleggghost, 2019-02-28:
I want to make a collision test:
A box falls freely to the ground, causing bouncing and rolling till stop, and we can check it on gepetto-viewer.
Should I start a new issue?
And any advices?

nmansard, 2019-02-28:
I started to change this tutorial to go to FCL detection collision. I then stopped because of some data resulting of FCL algorithms not properly bound to python-pinocchio. Related to #638

littleggghost, 2019-02-28:
Hi, @nmansard 
What do you think of my collision test as I mentioned above?

littleggghost, 2019-03-01:
I want to make a collision test:
A box falls freely to the ground, causing bouncing and rolling till stop, and we can check it on gepetto-viewer.
Should I start a new issue @jcarpent 
And any advices?

littleggghost, 2019-03-01:
Hi, @jmirabel 
I noticed you are familiar with FCL. Could you check this exercise?

jmirabel, 2019-03-05:
We have in mind to make hpp-fcl more robust (and unit-tested) but there is no timeline yet. I have no time to allocate to this. I could provide some guideline if anyone is willing to handle this.

olivier-stasse, 2019-03-05:
@littleggghost To make the story short, when starting hpp-fcl the team did not wanted to redo another collision library. But all the PR to fcl were ignored forcing us to do a fork. When the fcl library was reboot it was mostly to work on the CPP interface not on correcting the algorithms. @florent-lamiraux could give a more accurate account on this.

jcarpent, 2019-03-08:
This issue has been fixed by #725.

closed 2019-03-08 by jcarpent