stack-of-tasks/pinocchio #707: Error about createData() in practical-exercises?

littleggghost, 2019-03-01:
Hi, all
I do the practical-exercises in [1.1) Creating and displaying the robot](http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-practical-exercises_1-directgeom.html)
I meet an error:
`Traceback (most recent call last):
  File "exercise_1_1.py", line 29, in <module>
    robot = RobotWrapper(URDF, [PKG])
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 52, in __init__
    self.data = self.model.createData()
AttributeError: 'str' object has no attribute 'createData'`
Maybe the robot_wrapper.py is the old version? I'm sure I have compiled the v2.0.0 and installed it successfully.

jcarpent, 2019-03-01:
Yes, the code needs to be updated according to the new signature of the RobotWrapper constructor.

littleggghost, 2019-03-01:
Then where can I find the newest code?

jcarpent, 2019-03-01:
We just need to update the exercises. Can you do it please?

littleggghost, 2019-03-01:
I'm tring to do this. While I learn how to use pinocchio, I meet quite much errors. 
I wish the bug issues can be solved so that I can proceed.

littleggghost, 2019-03-01:
For this error, where should I start? or who could I contact that in charge of this before?

littleggghost, 2019-03-01:
Or where to find the related function of `self.model`? It seems the functions are under libpinocchio_pywrap.so file.

littleggghost, 2019-03-01:
I got it. The bug is right for v1.3.3, but for v2.0.0, it should be:
```
model_path = "the path to ur5"
mesh_dir = model_path
urdf_filename = "ur5_gripper.urdf"
urdf_model_path = model_path + "/ur5_description/urdf/" + urdf_filename
robot = RobotWrapper.BuildFromURDF(urdf_model_path, [mesh_dir], se3.JointModelFreeFlyer())
```

littleggghost, 2019-03-01:
Should this correction make by you or a pull request by me?

littleggghost, 2019-03-01:
By the way, there are still some changes should make in `ur5_gripper.urdf` and `ur5.urdf` files.
`ur_description` should be replaced as `ur5_description`.

jcarpent, 2019-03-01:
You may open a PR and I will review it. It will avoid being overloaded for us.

jcarpent, 2019-03-05:
@littleggghost Can you provide the fix you suggested last week?

littleggghost, 2019-03-06:
I noticed that this changes belong to `robotpkg-ur5-description`, and I should't make a PR here.
I post it here just for reference.
[urdf.zip](https://github.com/stack-of-tasks/pinocchio/files/2934176/urdf.zip)

closed 2019-03-06 by littleggghost