stack-of-tasks/pinocchio #708: Segmentation fault when using computeCollision() or computeCollisions()?

littleggghost, 2019-03-01:
Hi
While I do this  [5.0) prerequisites](http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-practical-exercises_5-planner.html) in python,  like calling:
`se3.computeCollision(robot.collision_model, robot.collision_data, 2)` or
`se3.computeCollisions(robot.collision_model, robot.collision_data, False)`
the program only crashed with `Segmentation fault (core dumped)`. 
I even don't know what happens. 
And should two objects bounce off when collision happens?

jcarpent, 2019-03-01:
Good catch. I will work on it next week. Thanks @littleggghost for this nice and helpful feedback.

jcarpent, 2019-03-05:
@littleggghost I have pushed a new Python example doing collision checking. Can you try it?
See this README: https://github.com/stack-of-tasks/pinocchio/tree/devel/examples/python

littleggghost, 2019-03-05:
After a little modify for v2.0.0, it meets an error:
```
  File "collision_detection.py", line 23, in <module>
    geom_model = pin.buildGeomFromUrdf(model,urdf_model_path,pin.GeometryType.COLLISION)
ValueError: Mesh package://romeo_description/meshes/V1/collision/TrunkYaw.dae could not be found.
```
I'm sure the dae file is exists.

littleggghost, 2019-03-05:
By the way, it will be better if it can add a ground collision box and be visualized on gepetto-viewer.

jcarpent, 2019-03-05:
It is very hard to identify such a bug, as it is working on a clean installation.
can you show that the file exists?
By the way, I won't be able to provide additional help. I think this is the max that I can do remotely.

jcarpent, 2019-03-05:
> By the way, it will be better if it can add a ground collision box and be visualized on gepetto-viewer.

Sorry, but I cannot provide more help for this feature. The only thing I can tell you is I'm willing to help you build such an example.

littleggghost, 2019-03-05:
You can check this pic:
[image]

jcarpent, 2019-03-05:
You've commented line 17  the correct line and put another one.

littleggghost, 2019-03-05:
Or I will get:
`Traceback (most recent call last):
  File "collision-detection.py", line 17, in <module>
    geom_model = pin.buildGeomFromUrdf(model,urdf_model_path,[model_path],pin.GeometryType.COLLISION)
Boost.Python.ArgumentError: Python argument types in
    pinocchio.libpinocchio_pywrap.buildGeomFromUrdf(Model, str, list, GeometryType)
did not match C++ signature:
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to assosiate the Geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > filename (string), pinocchio::GeometryType Geometry type, boost::shared_ptr<hpp::fcl::MeshLoader> Mesh loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Model to assosiate the Geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > filename (string), std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dirs (vector of strings), pinocchio::GeometryType Geometry type, boost::shared_ptr<hpp::fcl::MeshLoader> Mesh loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > Model to assosiate the Geometry, pinocchio::GeometryType filename (string))
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > Model to assosiate the Geometry, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > filename (string), pinocchio::GeometryType package_dirs (vector of strings))
`

jcarpent, 2019-03-05:
Have you reinstalled pinocchio from devel?

littleggghost, 2019-03-05:
I install v2.1.0 today through robotpkg

jcarpent, 2019-03-05:
Ok. You can then replace `[model_path]` by `pin.utils.fromListToVectorOfString([model_path])`.
The current devel branch is in advance and it is now more flexible.

littleggghost, 2019-03-05:
I modified two lines here and it has no error now:
1. change `pin.removeCollisionPairs(model,geom_model,srdf_model_path)`
to             `pin.removeCollisionPairs(model,geom_model,srdf_model_path, True)`
2. change `pin.loadReferenceConfigurations(model,srdf_model_path)`
to             `pin.loadReferenceConfigurations(model,srdf_model_path, True)`
Is there some packages version is out of date?

littleggghost, 2019-03-05:
I will keep exploring pinocchio!

jcarpent, 2019-03-05:
You have to install the devel version, which supports this new signatures.

nim65s, 2019-03-05:
Or wait for a new release ;)

jcarpent, 2019-03-05:
I close this issue as it seems to be solved.

closed 2019-03-05 by jcarpent