stack-of-tasks/pinocchio #709: Mac Pinocchio install failure

ddliugit, 2019-03-04:
I would like to install Pinocchio on Mac. When I try to import Pinocchio in the python2, I met the following error:

ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN5boost6system16generic_categoryEv
Referenced from: /usr/local/lib/libhpp-fcl.dylib
Expected in: /usr/local/opt/boost/lib/libboost_system-mt.dylib
in /usr/local/lib/libhpp-fcl.dylib

I guess this is caused by the library of hpp-fcl or the boost, though I have tried many times to install these two packages with latest home-brew. Can you help me out?

jcarpent, 2019-03-04:
What is your Mac version?

jcarpent, 2019-03-04:
I have updated the brew formula for **pinocchio** and **hpp-fcl**. @ddliugit Can you tell me if everything goes fine now?

ddliugit, 2019-03-04:
Mac Mojave version:10.14.3 

Thanks, I will try it again

ddliugit, 2019-03-04:
I tried it just now and found the problem changed.

I guess there exists problem in hpp-fcl library.

The following is the log in my terminal.

Can you help me find the problem?

My Mac version is Mojave 10.14.3

$ python2
Python 2.7.16 (default, Mar  4 2019, 09:01:38) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from pinocchio.robot_wrapper import RobotWrapper
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN3fcl8BVHModelINS_6OBBRSSEE10beginModelEii
  Referenced from: /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
  Expected in: /usr/local/lib/libhpp-fcl.dylib
 in /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so

ddliugit, 2019-03-04:
Not solved 
https://github.com/stack-of-tasks/pinocchio/issues/709

jcarpent, 2019-03-04:
can you do `brew reinstall hpp-fcl`?

ddliugit, 2019-03-04:
Yes, I tried it just now

The problem still exists and this is the log in the system 

$ brew reinstall hpp-fcl
==> Reinstalling gepetto/gepetto/hpp-fcl 
==> Downloading https://github.com/humanoid-path-planner/hpp-fcl/releases/downlo
Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/6d8c8e604ca0a47222d1ff471ebaf21f0597701a7dd83742eb7e7e9338c8bf96--hpp-fcl-1.0.1.mojave.bottle.1.tar.gz
==> Pouring hpp-fcl-1.0.1.mojave.bottle.1.tar.gz
🍺  /usr/local/Cellar/hpp-fcl/1.0.1: 1,666 files, 13.9MB
Dongdongs-MacBook-Pro:~ dongdong$ python2
Python 2.7.16 (default, Mar  4 2019, 09:01:38) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from pinocchio.robot_wrapper import RobotWrapper
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN3fcl8BVHModelINS_6OBBRSSEE10beginModelEii
  Referenced from: /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
  Expected in: /usr/local/lib/libhpp-fcl.dylib
 in /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so

ddliugit, 2019-03-04:
What is this problem

Did you test in MAC Mojave

jcarpent, 2019-03-04:
Yes, I'm on Mojave. 
The problem is that you do not have the missing hpp::fcl namespace.

jcarpent, 2019-03-04:
Can you try to reinstall from source both hpp-fcl and pinocchio:
``` brew reinstall --build-from-source hpp-fcl```
and
``` brew reinstall --build-from-source pinocchio ```

jcarpent, 2019-03-04:
In fact, I think you need to reinstall pinocchio too. This is the problem here.

ddliugit, 2019-03-04:
I just tried to reinstall and build from source, but the same problem happened again 

The log info from the terminal:

$ python2
Python 2.7.16 (default, Mar  4 2019, 09:01:38) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from pinocchio.robot_wrapper import RobotWrapper
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN3fcl8BVHModelINS_6OBBRSSEE10beginModelEii
  Referenced from: /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
  Expected in: /usr/local/lib/libhpp-fcl.dylib
 in /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so

ddliugit, 2019-03-04:
Does it work well on your Mac Mojave?

$ brew reinstall --build-from-source hpp-fcl
==> Reinstalling gepetto/gepetto/hpp-fcl 
==> Downloading https://github.com/humanoid-path-planner/hpp-fcl/releases/downlo
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws
######################################################################## 100.0%
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG
==> make
==> make test
==> make install
🍺  /usr/local/Cellar/hpp-fcl/1.0.1: 833 files, 10.6MB, built in 2 minutes 21 seconds
Dongdongs-MacBook-Pro:~ dongdong$ brew reinstall --build-from-source pinocchio
==> Reinstalling gepetto/gepetto/pinocchio 
==> Downloading https://github.com/stack-of-tasks/pinocchio/releases/download/v2
Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/b54092714b1ba96c6aa78ae3a09f0223fcd1c8f2753c536f01b2e0bddc3079fa--pinocchio-2.1.0.tar.gz
==> Patching
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG
==> make
==> make install
Warning: gepetto/gepetto/pinocchio dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink include/pinocchio/algorithm/aba-derivatives.hpp
Target /usr/local/include/pinocchio/algorithm/aba-derivatives.hpp
already exists. You may want to remove it:
  rm '/usr/local/include/pinocchio/algorithm/aba-derivatives.hpp'

To force the link and overwrite all conflicting files:
  brew link --overwrite pinocchio

To list all files that would be deleted:
  brew link --overwrite --dry-run pinocchio

Possible conflicting files are:
/usr/local/include/pinocchio/algorithm/aba-derivatives.hpp
/usr/local/include/pinocchio/algorithm/aba-derivatives.hxx
/usr/local/include/pinocchio/algorithm/aba.hpp
/usr/local/include/pinocchio/algorithm/aba.hxx
/usr/local/include/pinocchio/algorithm/center-of-mass.hpp
/usr/local/include/pinocchio/algorithm/center-of-mass.hxx
/usr/local/include/pinocchio/algorithm/centroidal-derivatives.hpp
/usr/local/include/pinocchio/algorithm/centroidal-derivatives.hxx
/usr/local/include/pinocchio/algorithm/centroidal.hpp
/usr/local/include/pinocchio/algorithm/centroidal.hxx
/usr/local/include/pinocchio/algorithm/check.hpp
/usr/local/include/pinocchio/algorithm/check.hxx
/usr/local/include/pinocchio/algorithm/cholesky.hpp
/usr/local/include/pinocchio/algorithm/cholesky.hxx
/usr/local/include/pinocchio/algorithm/compute-all-terms.hpp
/usr/local/include/pinocchio/algorithm/copy.hpp
/usr/local/include/pinocchio/algorithm/crba.hpp
/usr/local/include/pinocchio/algorithm/crba.hxx
/usr/local/include/pinocchio/algorithm/default-check.hpp
/usr/local/include/pinocchio/algorithm/dynamics.hpp
/usr/local/include/pinocchio/algorithm/energy.hpp
/usr/local/include/pinocchio/algorithm/finite-differences.hpp
/usr/local/include/pinocchio/algorithm/finite-differences.hxx
/usr/local/include/pinocchio/algorithm/frames.hpp
/usr/local/include/pinocchio/algorithm/frames.hxx
/usr/local/include/pinocchio/algorithm/geometry.hpp
/usr/local/include/pinocchio/algorithm/geometry.hxx
/usr/local/include/pinocchio/algorithm/jacobian.hpp
/usr/local/include/pinocchio/algorithm/jacobian.hxx
/usr/local/include/pinocchio/algorithm/joint-configuration.hpp
/usr/local/include/pinocchio/algorithm/joint-configuration.hxx
/usr/local/include/pinocchio/algorithm/kinematics-derivatives.hpp
/usr/local/include/pinocchio/algorithm/kinematics-derivatives.hxx
/usr/local/include/pinocchio/algorithm/kinematics.hpp
[... 1789 lines cut ...]
/usr/local/lib/python2.7/site-packages/pinocchio/explog.py
/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py
/usr/local/lib/python2.7/site-packages/pinocchio/romeo_wrapper.py
/usr/local/lib/python2.7/site-packages/pinocchio/rpy.py
/usr/local/lib/python2.7/site-packages/pinocchio/shortcuts.py
/usr/local/lib/python2.7/site-packages/pinocchio/utils.py
==> Summary
🍺  /usr/local/Cellar/pinocchio/2.1.0: 1,845 files, 31.2MB, built in 8 minutes 49 seconds
Dongdongs-MacBook-Pro:~ dongdong$ python2
Python 2.7.16 (default, Mar  4 2019, 09:01:38) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from pinocchio.robot_wrapper import RobotWrapper
  File "/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __ZN3fcl8BVHModelINS_6OBBRSSEE10beginModelEii
  Referenced from: /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
  Expected in: /usr/local/lib/libhpp-fcl.dylib
 in /usr/local/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
>>>

jcarpent, 2019-03-04:
As written in the log, you forgot to `brew link --overwrite pinocchio`. Please do it and try it again.

ddliugit, 2019-03-04:
Yes, it worked!!

Thanks a lot.

closed 2019-03-04 by ddliugit