stack-of-tasks/pinocchio #710: Tutorial link cannot be opened

ddliugit, 2019-03-04:
On the website(https://stack-of-tasks.github.io/pinocchio/) the tutorial cannot be opened 

See this link http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-labs_1-directgeom.html

nim65s, 2019-03-04:
Thanks for the report, I will fix this shortly

ddliugit, 2019-03-04:
Is there a walking tutorial about Pinocchio?

jcarpent, 2019-03-05:
The walking tutorials are old now. They will need a serious refresh.

ddliugit, 2019-03-05:
Are these tutorials in the repository?

I just would to calculate the M inertia matrix in the equtation

https://latex.codecogs.com/gif.latex?M%28%5Ctheta%29%5Cddot%7B%5Ctheta%7D&plus;%20C%28%5Ctheta%2C%5Cdot%7B%5Ctheta%7D%29%5Cdot%7B%5Ctheta%7D&plus;G%28%5Ctheta%29%3D%20S%5ET%5Ctau&plus;J_c%28%5Ctheta%29f_c

Is there a function which can calculate this matrix after importing the Urdf model in Pinocchio?

jcarpent, 2019-03-05:
You can use:
```
data = model.createData()
pin.crba(model,data,q) 
```
where `q` is your configuration vector.

gabrielebndn, 2019-03-05:
Hi @ddliugit, can you check whether the link is working now?

jcarpent, 2019-03-07:
@ddliugit I think the problem is now solved. To no hesitate to reopen this issue if needed.

closed 2019-03-07 by jcarpent

ddliugit, 2019-03-19:
The tutorial is closed. Can you open it

ddliugit, 2019-03-19:
The website is closed.

nim65s, 2019-03-19:
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_1-directgeom.html seems OK for me.

nim65s, 2019-03-19:
Oh, your link was wrong, and I fixed it without noticing, sorry.