stack-of-tasks/pinocchio #713: Pinocchio Mac Model importation problem

ddliugit, 2019-03-04:
1. Does robot wrapper support model importation on Mac?

2. Also, can Mac user use the graphic interface to show the robot motion on Mac like in ubuntu? 

I tried to install gepetto-viewer  on Mac, but it fails to compile. 

Can you help me find the problem 

In [1]: from pinocchio.robot_wrapper import RobotWrapper
   ...: 
   ...: pkg  = '$root_vol$HOME/Desktop/models'
   ...: urdf = pkg + '/ur_description/urdf/ur5_gripper.urdf'
   ...: robot = RobotWrapper(urdf,[pkg,])
   ...: 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-3f0456183d71> in <module>()
      3 pkg  = '$root_vol$HOME/Desktop/models'
      4 urdf = pkg + '/ur_description/urdf/ur5_gripper.urdf'
----> 5 robot = RobotWrapper(urdf,[pkg,])

/usr/local/lib/python2.7/site-packages/pinocchio/robot_wrapper.pyc in __init__(self, model, collision_model, visual_model, verbose)
     64 
     65         self.model = model
---> 66         self.data = self.model.createData()
     67 
     68         self.collision_model = collision_model

AttributeError: 'str' object has no attribute 'createData'

jcarpent, 2019-03-05:
To install gepetto-viewer, you just need to do 
```brew reinstall gepetto-viewer-corba```.
For the rest, it is related to #707.

jcarpent, 2019-03-05:
@ddliugit Where have you found these instructions to use RobotWrapper?

gabrielebndn, 2019-03-05:
Dear @ddliugit, are you installing Pinocchio 2.0, from source? In this case, the constructor RobotWrapper(urdf) is outdated. For a similar behavior, you should use the static method RobotWrapper.BuildFromURDF

jcarpent, 2019-03-05:
@gabrielebndn Can you update the related examples and tutorials?

ddliugit, 2019-03-05:
@jcarpent followed your instructions and installed the gepetto-viewer-corba, but do you know how to start it like the command in the tutorial? "gepetto-gui" doesn't work any more

@gabrielebndn I followed your instructions to use build from URDF, it worked well!!
Thanks a lot!
In the official website you mentioned there is 20 hours of tutorial about walking, can you tell me where it is?

jcarpent, 2019-03-05:
What do you mean by does not work anymore?

ddliugit, 2019-03-05:
I just followed the tutorial and would like to show the robot movement
$ gepetto-gui
-bash: gepetto-gui: command not found

jcarpent, 2019-03-05:
Which tutorial? and how do you install gepetto-viewer and so gepetto-gui?

ddliugit, 2019-03-05:
http://homepages.laas.fr/nmansard/teach/robotics2018/tp1.html

I followed your instruction:  brew reinstall gepetto-viewer-corba

jcarpent, 2019-03-05:
gepetto-gui is located there: `/usr/local/opt/gepetto-viewer-corba/bin`
Can you show the content of this directory?

ddliugit, 2019-03-05:
I think I installed it, but I cannot find this directory

$ cd /usr/local/opt/gepetto-viewer-corba/bin
-bash: cd: /usr/local/opt/gepetto-viewer-corba/bin: No such file or directory

ddliugit, 2019-03-05:
I will this command again:brew reinstall gepetto-viewer-corba

ddliugit, 2019-03-05:
The content is:
$cd /usr/local/opt/gepetto-viewer-corba/bin
$ ls
gepetto-gui		gepetto-viewer-server	urdf_to_blender.py

ddliugit, 2019-03-05:
Oh, I think I didn't install it well before, now the problem changed

$ gepetto-gui
dyld: Library not loaded: @@HOMEBREW_PREFIX@@/opt/open-scene-graph/lib/libosg.157.dylib
  Referenced from: /usr/local/bin/gepetto-gui
  Reason: image not found
Abort trap: 6

jcarpent, 2019-03-05:
I will fix this issue soon. I know the reason.

closed 2019-03-05 by jcarpent

ddliugit, 2019-03-06:
Sorry, I am still confused about this problem. Did you fix it?

ddliugit, 2019-03-06:
I tried to reinstall the problem, but this error appears 

$ brew reinstall gepetto-viewer-corba
Error: No available formula with the name "gepetto/gepetto/open-scene-graph-with-colladadom" (dependency of gepetto/gepetto/gepetto-viewer-corba)

jcarpent, 2019-03-06:
I was closing this issue because the rest was not related to Pinocchio.
I fixed the issue now, you can proceed as follows:
```brew install gepetto-viewer-corba```

ddliugit, 2019-03-06:
I found this problem when I tried to install it 

$ brew install gepetto-viewer-corba
==> Installing gepetto-viewer-corba from gepetto/gepetto
Error: No available formula with the name "gepetto/gepetto/open-scene-graph-with-colladadom" (dependency of gepetto/gepetto/gepetto-viewer-corba)
==> Searching for a previously deleted formula (in the last month)...
Warning: gepetto/gepetto is shallow clone. To get complete history run:
  git -C "$(brew --repo gepetto/gepetto)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.

jcarpent, 2019-03-06:
Sorry, I have forgotten to publish the missing formula.
Can you try it again?

ddliugit, 2019-03-06:
I tried it and problem is like this:

$ brew install gepetto-viewer-corba
==> Installing gepetto-viewer-corba from gepetto/gepetto
==> Downloading https://github.com/humanoid-path-planner/gepetto-viewer-corba/re
Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/9aabd3039d6aa91690e3b631b5da7160858b2b77716bceef1be214a9429cabca--gepetto-viewer-corba-5.1.3.mojave.bottle.tar.gz
==> Pouring gepetto-viewer-corba-5.1.3.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/gepetto-viewer-corba/5.1.3: 1,216 files, 6.4MB

$ gepetto-gui
dyld: Library not loaded: /usr/local/opt/osgqt/lib/libosgQt5.145.dylib
  Referenced from: /usr/local/bin/gepetto-gui
  Reason: image not found
Abort trap: 6

ddliugit, 2019-03-06:
Same problem? Did I install the wrong package

ddliugit, 2019-03-06:
I found reading dae model error in this Gui:

$ gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
qt.qpa.window: Window position QRect(208,-55 1023x1009) outside any known screen, using primary screen
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
   hpp-gui
   python-pinocchio
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/base.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/base.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/base.dae Desktop/models/ur_description/meshes/ur5/visual/base.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/base.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/forearm.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/forearm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/forearm.dae Desktop/models/ur_description/meshes/ur5/visual/forearm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/forearm.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae found but could not be opened. Check that a plugin exist.
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/visuals/base_link_0" does not exist.
Node "world/pinocchio/visuals/shoulder_link_0" does not exist.
Node "world/pinocchio/visuals/upper_arm_link_0" does not exist.
Node "world/pinocchio/visuals/forearm_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_1_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_2_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_3_link_0" does not exist.
Node "world/pinocchio/visuals/tool0_0" does not exist.
Node "world/pinocchio/visuals/tool0_1" does not exist.
Node "world/pinocchio/visuals/tool0_2" does not exist.
()

jcarpent, 2019-03-06:
Can you do:
```brew unlink open-scene-graph```
then:
```brew link open-scene-graph-with-colladadom```

ddliugit, 2019-03-06:
I tried, the error is like this after loading the model:

gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
qt.qpa.window: Window position QRect(208,-55 1023x1009) outside any known screen, using primary screen
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
   hpp-gui
   python-pinocchio
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/base.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/base.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/base.dae Desktop/models/ur_description/meshes/ur5/visual/base.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/base.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/forearm.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/forearm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/forearm.dae Desktop/models/ur_description/meshes/ur5/visual/forearm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/forearm.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/wrist2.dae found but could not be opened. Check that a plugin exist.
Load failed in COLLADA DOM
Load failed in COLLADA DOM conversion
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae: read error
File: Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/wrist3.dae found but could not be opened. Check that a plugin exist.
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/visuals/base_link_0" does not exist.
Node "world/pinocchio/visuals/shoulder_link_0" does not exist.
Node "world/pinocchio/visuals/upper_arm_link_0" does not exist.
Node "world/pinocchio/visuals/forearm_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_1_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_2_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_3_link_0" does not exist.
Node "world/pinocchio/visuals/tool0_0" does not exist.
Node "world/pinocchio/visuals/tool0_1" does not exist.
Node "world/pinocchio/visuals/tool0_2" does not exist.

jcarpent, 2019-03-06:
Is `/usr/local/lib` inside your LD_LIBRARY_PATH?

ddliugit, 2019-03-06:
Yes
This is my .bash_profile
export PATH=/usr/local/bin:$PATH
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/usr/local/lib/python2.7/site-packages:$PYTHONPATH

jcarpent, 2019-03-06:
Then, can you show the content of:
```ls -l /usr/local/opt/open-scene-graph-with-colladadom/lib/osgPlugins-3.6.3/```
and
```ls -l /usr/local/lib/osgPlugins-3.6.3/```

ddliugit, 2019-03-06:
$ ls -l /usr/local/opt/open-scene-graph-with-colladadom/lib/osgPlugins-3.6.3/
total 33800
-r--r--r--  1 dongdong  staff    73032 Mar  6 08:50 osgdb_3dc.so
-r--r--r--  1 dongdong  staff   311744 Mar  6 08:50 osgdb_3ds.so
-r--r--r--  1 dongdong  staff   176988 Mar  6 08:50 osgdb_ac.so
-r--r--r--  1 dongdong  staff   127188 Mar  6 08:50 osgdb_avfoundation.so
-r--r--r--  1 dongdong  staff    49424 Mar  6 08:50 osgdb_bmp.so
-r--r--r--  1 dongdong  staff   242616 Mar  6 08:50 osgdb_bsp.so
-r--r--r--  1 dongdong  staff   124616 Mar  6 08:50 osgdb_bvh.so
-r--r--r--  1 dongdong  staff   161368 Mar  6 08:50 osgdb_cfg.so
-r--r--r--  1 dongdong  staff   108888 Mar  6 08:50 osgdb_curl.so
-r--r--r--  1 dongdong  staff   848864 Mar  6 08:50 osgdb_dae.so
-r--r--r--  1 dongdong  staff    74468 Mar  6 08:50 osgdb_dds.so
-r--r--r--  1 dongdong  staff   651384 Mar  6 08:50 osgdb_deprecated_osg.so
-r--r--r--  1 dongdong  staff   215956 Mar  6 08:50 osgdb_deprecated_osganimation.so
-r--r--r--  1 dongdong  staff    46048 Mar  6 08:50 osgdb_deprecated_osgfx.so
-r--r--r--  1 dongdong  staff   180676 Mar  6 08:50 osgdb_deprecated_osgparticle.so
-r--r--r--  1 dongdong  staff    30240 Mar  6 08:50 osgdb_deprecated_osgshadow.so
-r--r--r--  1 dongdong  staff    96960 Mar  6 08:50 osgdb_deprecated_osgsim.so
-r--r--r--  1 dongdong  staff    77344 Mar  6 08:50 osgdb_deprecated_osgterrain.so
-r--r--r--  1 dongdong  staff    62208 Mar  6 08:50 osgdb_deprecated_osgtext.so
-r--r--r--  1 dongdong  staff    44700 Mar  6 08:50 osgdb_deprecated_osgviewer.so
-r--r--r--  1 dongdong  staff    72364 Mar  6 08:50 osgdb_deprecated_osgvolume.so
-r--r--r--  1 dongdong  staff    69372 Mar  6 08:50 osgdb_deprecated_osgwidget.so
-r--r--r--  1 dongdong  staff    79584 Mar  6 08:50 osgdb_dot.so
-r--r--r--  1 dongdong  staff   246156 Mar  6 08:50 osgdb_dxf.so
-r--r--r--  1 dongdong  staff    87128 Mar  6 08:50 osgdb_freetype.so
-r--r--r--  1 dongdong  staff  1006632 Mar  6 08:50 osgdb_gles.so
-r--r--r--  1 dongdong  staff    62496 Mar  6 08:50 osgdb_glsl.so
-r--r--r--  1 dongdong  staff    69296 Mar  6 08:50 osgdb_gz.so
-r--r--r--  1 dongdong  staff    63168 Mar  6 08:50 osgdb_hdr.so
-r--r--r--  1 dongdong  staff    81240 Mar  6 08:50 osgdb_imageio.so
-r--r--r--  1 dongdong  staff   751308 Mar  6 08:50 osgdb_ive.so
-r--r--r--  1 dongdong  staff    49084 Mar  6 08:50 osgdb_jp2.so
-r--r--r--  1 dongdong  staff    67272 Mar  6 08:50 osgdb_jpeg.so
-r--r--r--  1 dongdong  staff    61296 Mar  6 08:50 osgdb_ktx.so
-r--r--r--  1 dongdong  staff    84100 Mar  6 08:50 osgdb_logo.so
-r--r--r--  1 dongdong  staff   543100 Mar  6 08:50 osgdb_lua.so
-r--r--r--  1 dongdong  staff   325236 Mar  6 08:50 osgdb_lwo.so
-r--r--r--  1 dongdong  staff    96164 Mar  6 08:50 osgdb_lws.so
-r--r--r--  1 dongdong  staff    74760 Mar  6 08:50 osgdb_md2.so
-r--r--r--  1 dongdong  staff   111500 Mar  6 08:50 osgdb_mdl.so
-r--r--r--  1 dongdong  staff    99180 Mar  6 08:50 osgdb_normals.so
-r--r--r--  1 dongdong  staff   228080 Mar  6 08:50 osgdb_obj.so
-r--r--r--  1 dongdong  staff   538416 Mar  6 08:50 osgdb_openflight.so
-r--r--r--  1 dongdong  staff   247468 Mar  6 08:50 osgdb_osc.so
-r--r--r--  1 dongdong  staff   179264 Mar  6 08:50 osgdb_osg.so
-r--r--r--  1 dongdong  staff   110836 Mar  6 08:50 osgdb_osga.so
-r--r--r--  1 dongdong  staff   434768 Mar  6 08:50 osgdb_osgjs.so
-r--r--r--  1 dongdong  staff    43636 Mar  6 08:50 osgdb_osgshadow.so
-r--r--r--  1 dongdong  staff    56640 Mar  6 08:50 osgdb_osgterrain.so
-r--r--r--  1 dongdong  staff    44960 Mar  6 08:50 osgdb_osgtgz.so
-r--r--r--  1 dongdong  staff    60048 Mar  6 08:50 osgdb_osgviewer.so
-r--r--r--  1 dongdong  staff   313860 Mar  6 08:50 osgdb_p3d.so
-r--r--r--  1 dongdong  staff    37116 Mar  6 08:50 osgdb_pic.so
-r--r--r--  1 dongdong  staff   110748 Mar  6 08:50 osgdb_ply.so
-r--r--r--  1 dongdong  staff    67120 Mar  6 08:50 osgdb_pnm.so
-r--r--r--  1 dongdong  staff   119280 Mar  6 08:50 osgdb_pov.so
-r--r--r--  1 dongdong  staff    53912 Mar  6 08:50 osgdb_pvr.so
-r--r--r--  1 dongdong  staff    52848 Mar  6 08:50 osgdb_revisions.so
-r--r--r--  1 dongdong  staff    50300 Mar  6 08:50 osgdb_rgb.so
-r--r--r--  1 dongdong  staff    38884 Mar  6 08:50 osgdb_rot.so
-r--r--r--  1 dongdong  staff    39204 Mar  6 08:50 osgdb_scale.so
-r--r--r--  1 dongdong  staff    51432 Mar  6 08:50 osgdb_sdl.so
-r--r--r--  1 dongdong  staff  1937484 Mar  6 08:50 osgdb_serializers_osg.so
-r--r--r--  1 dongdong  staff   715540 Mar  6 08:50 osgdb_serializers_osganimation.so
-r--r--r--  1 dongdong  staff    87928 Mar  6 08:50 osgdb_serializers_osgfx.so
-r--r--r--  1 dongdong  staff   129588 Mar  6 08:50 osgdb_serializers_osgga.so
-r--r--r--  1 dongdong  staff   126436 Mar  6 08:50 osgdb_serializers_osgmanipulator.so
-r--r--r--  1 dongdong  staff   345260 Mar  6 08:50 osgdb_serializers_osgparticle.so
-r--r--r--  1 dongdong  staff   185352 Mar  6 08:50 osgdb_serializers_osgshadow.so
-r--r--r--  1 dongdong  staff   239848 Mar  6 08:50 osgdb_serializers_osgsim.so
-r--r--r--  1 dongdong  staff   175244 Mar  6 08:50 osgdb_serializers_osgterrain.so
-r--r--r--  1 dongdong  staff   145412 Mar  6 08:50 osgdb_serializers_osgtext.so
-r--r--r--  1 dongdong  staff   255764 Mar  6 08:50 osgdb_serializers_osgui.so
-r--r--r--  1 dongdong  staff    22664 Mar  6 08:50 osgdb_serializers_osgutil.so
-r--r--r--  1 dongdong  staff    92268 Mar  6 08:50 osgdb_serializers_osgviewer.so
[... cut ...]
-r--r--r--  1 dongdong  staff    38892 Mar  6 08:50 osgdb_trans.so
-r--r--r--  1 dongdong  staff    72660 Mar  6 08:50 osgdb_trk.so
-r--r--r--  1 dongdong  staff    52516 Mar  6 08:50 osgdb_txf.so
-r--r--r--  1 dongdong  staff   731968 Mar  6 08:50 osgdb_txp.so
-r--r--r--  1 dongdong  staff    54712 Mar  6 08:50 osgdb_vtf.so
-r--r--r--  1 dongdong  staff   115544 Mar  6 08:50 osgdb_x.so
-r--r--r--  1 dongdong  staff    92448 Mar  6 08:50 osgdb_zeroconf.so
-r--r--r--  1 dongdong  staff   140076 Mar  6 08:50 osgdb_zip.so

ddliugit, 2019-03-06:
$ ls -l /usr/local/lib/osgPlugins-3.6.3/
total 33800
-r--r--r--  1 dongdong  staff    73032 Mar  6 08:50 osgdb_3dc.so
-r--r--r--  1 dongdong  staff   311744 Mar  6 08:50 osgdb_3ds.so
-r--r--r--  1 dongdong  staff   176988 Mar  6 08:50 osgdb_ac.so
-r--r--r--  1 dongdong  staff   127188 Mar  6 08:50 osgdb_avfoundation.so
-r--r--r--  1 dongdong  staff    49424 Mar  6 08:50 osgdb_bmp.so
-r--r--r--  1 dongdong  staff   242616 Mar  6 08:50 osgdb_bsp.so
-r--r--r--  1 dongdong  staff   124616 Mar  6 08:50 osgdb_bvh.so
-r--r--r--  1 dongdong  staff   161368 Mar  6 08:50 osgdb_cfg.so
-r--r--r--  1 dongdong  staff   108888 Mar  6 08:50 osgdb_curl.so
-r--r--r--  1 dongdong  staff   848864 Mar  6 08:50 osgdb_dae.so
-r--r--r--  1 dongdong  staff    74468 Mar  6 08:50 osgdb_dds.so
-r--r--r--  1 dongdong  staff   651384 Mar  6 08:50 osgdb_deprecated_osg.so
-r--r--r--  1 dongdong  staff   215956 Mar  6 08:50 osgdb_deprecated_osganimation.so
-r--r--r--  1 dongdong  staff    46048 Mar  6 08:50 osgdb_deprecated_osgfx.so
-r--r--r--  1 dongdong  staff   180676 Mar  6 08:50 osgdb_deprecated_osgparticle.so
-r--r--r--  1 dongdong  staff    30240 Mar  6 08:50 osgdb_deprecated_osgshadow.so
-r--r--r--  1 dongdong  staff    96960 Mar  6 08:50 osgdb_deprecated_osgsim.so
-r--r--r--  1 dongdong  staff    77344 Mar  6 08:50 osgdb_deprecated_osgterrain.so
-r--r--r--  1 dongdong  staff    62208 Mar  6 08:50 osgdb_deprecated_osgtext.so
-r--r--r--  1 dongdong  staff    44700 Mar  6 08:50 osgdb_deprecated_osgviewer.so
-r--r--r--  1 dongdong  staff    72364 Mar  6 08:50 osgdb_deprecated_osgvolume.so
-r--r--r--  1 dongdong  staff    69372 Mar  6 08:50 osgdb_deprecated_osgwidget.so
-r--r--r--  1 dongdong  staff    79584 Mar  6 08:50 osgdb_dot.so
-r--r--r--  1 dongdong  staff   246156 Mar  6 08:50 osgdb_dxf.so
-r--r--r--  1 dongdong  staff    87128 Mar  6 08:50 osgdb_freetype.so
-r--r--r--  1 dongdong  staff  1006632 Mar  6 08:50 osgdb_gles.so
-r--r--r--  1 dongdong  staff    62496 Mar  6 08:50 osgdb_glsl.so
-r--r--r--  1 dongdong  staff    69296 Mar  6 08:50 osgdb_gz.so
-r--r--r--  1 dongdong  staff    63168 Mar  6 08:50 osgdb_hdr.so
-r--r--r--  1 dongdong  staff    81240 Mar  6 08:50 osgdb_imageio.so
-r--r--r--  1 dongdong  staff   751308 Mar  6 08:50 osgdb_ive.so
-r--r--r--  1 dongdong  staff    49084 Mar  6 08:50 osgdb_jp2.so
-r--r--r--  1 dongdong  staff    67272 Mar  6 08:50 osgdb_jpeg.so
-r--r--r--  1 dongdong  staff    61296 Mar  6 08:50 osgdb_ktx.so
-r--r--r--  1 dongdong  staff    84100 Mar  6 08:50 osgdb_logo.so
-r--r--r--  1 dongdong  staff   543100 Mar  6 08:50 osgdb_lua.so
-r--r--r--  1 dongdong  staff   325236 Mar  6 08:50 osgdb_lwo.so
-r--r--r--  1 dongdong  staff    96164 Mar  6 08:50 osgdb_lws.so
-r--r--r--  1 dongdong  staff    74760 Mar  6 08:50 osgdb_md2.so
-r--r--r--  1 dongdong  staff   111500 Mar  6 08:50 osgdb_mdl.so
-r--r--r--  1 dongdong  staff    99180 Mar  6 08:50 osgdb_normals.so
-r--r--r--  1 dongdong  staff   228080 Mar  6 08:50 osgdb_obj.so
-r--r--r--  1 dongdong  staff   538416 Mar  6 08:50 osgdb_openflight.so
-r--r--r--  1 dongdong  staff   247468 Mar  6 08:50 osgdb_osc.so
-r--r--r--  1 dongdong  staff   179264 Mar  6 08:50 osgdb_osg.so
-r--r--r--  1 dongdong  staff   110836 Mar  6 08:50 osgdb_osga.so
-r--r--r--  1 dongdong  staff   434768 Mar  6 08:50 osgdb_osgjs.so
-r--r--r--  1 dongdong  staff    43636 Mar  6 08:50 osgdb_osgshadow.so
-r--r--r--  1 dongdong  staff    56640 Mar  6 08:50 osgdb_osgterrain.so
-r--r--r--  1 dongdong  staff    44960 Mar  6 08:50 osgdb_osgtgz.so
-r--r--r--  1 dongdong  staff    60048 Mar  6 08:50 osgdb_osgviewer.so
-r--r--r--  1 dongdong  staff   313860 Mar  6 08:50 osgdb_p3d.so
-r--r--r--  1 dongdong  staff    37116 Mar  6 08:50 osgdb_pic.so
-r--r--r--  1 dongdong  staff   110748 Mar  6 08:50 osgdb_ply.so
-r--r--r--  1 dongdong  staff    67120 Mar  6 08:50 osgdb_pnm.so
-r--r--r--  1 dongdong  staff   119280 Mar  6 08:50 osgdb_pov.so
-r--r--r--  1 dongdong  staff    53912 Mar  6 08:50 osgdb_pvr.so
-r--r--r--  1 dongdong  staff    52848 Mar  6 08:50 osgdb_revisions.so
-r--r--r--  1 dongdong  staff    50300 Mar  6 08:50 osgdb_rgb.so
-r--r--r--  1 dongdong  staff    38884 Mar  6 08:50 osgdb_rot.so
-r--r--r--  1 dongdong  staff    39204 Mar  6 08:50 osgdb_scale.so
-r--r--r--  1 dongdong  staff    51432 Mar  6 08:50 osgdb_sdl.so
-r--r--r--  1 dongdong  staff  1937484 Mar  6 08:50 osgdb_serializers_osg.so
-r--r--r--  1 dongdong  staff   715540 Mar  6 08:50 osgdb_serializers_osganimation.so
-r--r--r--  1 dongdong  staff    87928 Mar  6 08:50 osgdb_serializers_osgfx.so
-r--r--r--  1 dongdong  staff   129588 Mar  6 08:50 osgdb_serializers_osgga.so
-r--r--r--  1 dongdong  staff   126436 Mar  6 08:50 osgdb_serializers_osgmanipulator.so
-r--r--r--  1 dongdong  staff   345260 Mar  6 08:50 osgdb_serializers_osgparticle.so
-r--r--r--  1 dongdong  staff   185352 Mar  6 08:50 osgdb_serializers_osgshadow.so
-r--r--r--  1 dongdong  staff   239848 Mar  6 08:50 osgdb_serializers_osgsim.so
-r--r--r--  1 dongdong  staff   175244 Mar  6 08:50 osgdb_serializers_osgterrain.so
-r--r--r--  1 dongdong  staff   145412 Mar  6 08:50 osgdb_serializers_osgtext.so
-r--r--r--  1 dongdong  staff   255764 Mar  6 08:50 osgdb_serializers_osgui.so
-r--r--r--  1 dongdong  staff    22664 Mar  6 08:50 osgdb_serializers_osgutil.so
-r--r--r--  1 dongdong  staff    92268 Mar  6 08:50 osgdb_serializers_osgviewer.so
-r--r--r--  1 dongdong  staff   227068 Mar  6 08:50 osgdb_serializers_osgvolume.so
[... cut ...]
-r--r--r--  1 dongdong  staff    38892 Mar  6 08:50 osgdb_trans.so
-r--r--r--  1 dongdong  staff    72660 Mar  6 08:50 osgdb_trk.so
-r--r--r--  1 dongdong  staff    52516 Mar  6 08:50 osgdb_txf.so
-r--r--r--  1 dongdong  staff   731968 Mar  6 08:50 osgdb_txp.so
-r--r--r--  1 dongdong  staff    54712 Mar  6 08:50 osgdb_vtf.so
-r--r--r--  1 dongdong  staff   115544 Mar  6 08:50 osgdb_x.so
-r--r--r--  1 dongdong  staff    92448 Mar  6 08:50 osgdb_zeroconf.so
-r--r--r--  1 dongdong  staff   140076 Mar  6 08:50 osgdb_zip.so

jcarpent, 2019-03-06:
`ls -l /usr/local/lib/osgPlugins-3.6.3/` ??

jcarpent, 2019-03-06:
Sorry. And this one `ls -l /usr/local/lib/`

ddliugit, 2019-03-06:
$ ls -l /usr/local/lib/
total 16584
lrwxr-xr-x    1 dongdong  admin       47 Mar  5 08:09 Qt3Support.framework -> ../Cellar/qt@4/4.8.7_5/lib/Qt3Support.framework
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:09 QtCore.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtCore.framework
lrwxr-xr-x    1 dongdong  admin       50 Mar  5 08:09 QtDeclarative.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtDeclarative.framework
lrwxr-xr-x    1 dongdong  admin       47 Mar  5 08:09 QtDesigner.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtDesigner.framework
lrwxr-xr-x    1 dongdong  admin       57 Mar  5 08:09 QtDesignerComponents.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtDesignerComponents.framework
lrwxr-xr-x    1 dongdong  admin       42 Mar  5 08:09 QtGui.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtGui.framework
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:09 QtHelp.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtHelp.framework
lrwxr-xr-x    1 dongdong  admin       49 Mar  5 08:09 QtMultimedia.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtMultimedia.framework
lrwxr-xr-x    1 dongdong  admin       46 Mar  5 08:09 QtNetwork.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtNetwork.framework
lrwxr-xr-x    1 dongdong  admin       45 Mar  5 08:09 QtOpenGL.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtOpenGL.framework
lrwxr-xr-x    1 dongdong  admin       45 Mar  5 08:09 QtScript.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtScript.framework
lrwxr-xr-x    1 dongdong  admin       50 Mar  5 08:09 QtScriptTools.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtScriptTools.framework
lrwxr-xr-x    1 dongdong  admin       42 Mar  5 08:09 QtSql.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtSql.framework
lrwxr-xr-x    1 dongdong  admin       42 Mar  5 08:09 QtSvg.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtSvg.framework
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:09 QtTest.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtTest.framework
lrwxr-xr-x    1 dongdong  admin       42 Mar  5 08:09 QtXml.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtXml.framework
lrwxr-xr-x    1 dongdong  admin       50 Mar  5 08:09 QtXmlPatterns.framework -> ../Cellar/qt@4/4.8.7_5/lib/QtXmlPatterns.framework
lrwxr-xr-x    1 dongdong  admin       32 Mar  5 08:08 cairo -> ../Cellar/cairo/1.16.0/lib/cairo
drwxr-xr-x    6 dongdong  admin      192 Mar  3 19:56 ccd
drwxr-xr-x    6 dongdong  admin      192 Mar  6 08:47 cmake
lrwxr-xr-x    1 dongdong  admin       49 Mar  1 13:59 console_bridge -> ../Cellar/console_bridge/0.4.3/lib/console_bridge
lrwxr-xr-x    1 dongdong  admin       27 Mar  1 13:59 gcc -> ../Cellar/gcc/8.3.0/lib/gcc
drwxr-xr-x    3 dongdong  admin       96 Mar  5 08:08 gdk-pixbuf-2.0
drwxr-xr-x    3 dongdong  admin       96 Mar  5 08:07 gio
drwxr-xr-x   11 dongdong  admin      352 Mar  5 08:08 girepository-1.0
lrwxr-xr-x    1 dongdong  admin       34 Mar  6 08:47 glib-2.0 -> ../Cellar/glib/2.60.0/lib/glib-2.0
lrwxr-xr-x    1 dongdong  admin       36 Mar  5 08:08 gtk-2.0 -> ../Cellar/gtk+/2.24.32_2/lib/gtk-2.0
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:08 gtkglext-1.0 -> ../Cellar/gtkglext/1.2.0_2/lib/gtkglext-1.0
lrwxr-xr-x    1 dongdong  admin       45 Mar  5 08:07 libCOS4.2.2.dylib -> ../Cellar/omniorb/4.2.2/lib/libCOS4.2.2.dylib
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:07 libCOS4.2.dylib -> ../Cellar/omniorb/4.2.2/lib/libCOS4.2.dylib
lrwxr-xr-x    1 dongdong  admin       37 Mar  5 08:07 libCOS4.a -> ../Cellar/omniorb/4.2.2/lib/libCOS4.a
lrwxr-xr-x    1 dongdong  admin       41 Mar  5 08:07 libCOS4.dylib -> ../Cellar/omniorb/4.2.2/lib/libCOS4.dylib
lrwxr-xr-x    1 dongdong  admin       52 Mar  5 08:07 libCOSDynamic4.2.2.dylib -> ../Cellar/omniorb/4.2.2/lib/libCOSDynamic4.2.2.dylib
lrwxr-xr-x    1 dongdong  admin       50 Mar  5 08:07 libCOSDynamic4.2.dylib -> ../Cellar/omniorb/4.2.2/lib/libCOSDynamic4.2.dylib
lrwxr-xr-x    1 dongdong  admin       44 Mar  5 08:07 libCOSDynamic4.a -> ../Cellar/omniorb/4.2.2/lib/libCOSDynamic4.a
lrwxr-xr-x    1 dongdong  admin       48 Mar  5 08:07 libCOSDynamic4.dylib -> ../Cellar/omniorb/4.2.2/lib/libCOSDynamic4.dylib
lrwxr-xr-x    1 dongdong  admin       38 Mar  1 14:00 libIrrXML.a -> ../Cellar/assimp/4.1.0/lib/libIrrXML.a
lrwxr-xr-x    1 dongdong  admin       76 Mar  6 09:56 libOpenThreads.21.dylib -> ../Cellar/open-scene-graph-with-colladadom/3.6.3/lib/libOpenThreads.21.dylib
lrwxr-xr-x    1 dongdong  admin       79 Mar  6 09:56 libOpenThreads.3.3.1.dylib -> ../Cellar/open-scene-graph-with-colladadom/3.6.3/lib/libOpenThreads.3.3.1.dylib
lrwxr-xr-x    1 dongdong  admin       73 Mar  6 09:56 libOpenThreads.dylib -> ../Cellar/open-scene-graph-with-colladadom/3.6.3/lib/libOpenThreads.dylib
lrwxr-xr-x    1 dongdong  admin       51 Mar  5 08:09 libQtCLucene.4.8.7.dylib -> ../Cellar/qt@4/4.8.7_5/lib/libQtCLucene.4.8.7.dylib
lrwxr-xr-x    1 dongdong  admin       49 Mar  5 08:09 libQtCLucene.4.8.dylib -> ../Cellar/qt@4/4.8.7_5/lib/libQtCLucene.4.8.dylib
lrwxr-xr-x    1 dongdong  admin       47 Mar  5 08:09 libQtCLucene.4.dylib -> ../Cellar/qt@4/4.8.7_5/lib/libQtCLucene.4.dylib
lrwxr-xr-x    1 dongdong  admin       45 Mar  5 08:09 libQtCLucene.dylib -> ../Cellar/qt@4/4.8.7_5/lib/libQtCLucene.dylib
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:09 libQtCLucene.prl -> ../Cellar/qt@4/4.8.7_5/lib/libQtCLucene.prl
lrwxr-xr-x    1 dongdong  admin       41 Mar  5 08:09 libQtUiTools.a -> ../Cellar/qt@4/4.8.7_5/lib/libQtUiTools.a
lrwxr-xr-x    1 dongdong  admin       43 Mar  5 08:09 libQtUiTools.prl -> ../Cellar/qt@4/4.8.7_5/lib/libQtUiTools.prl
[... cut ...]
drwxr-xr-x  132 dongdong  admin     4224 Mar  6 09:56 pkgconfig
drwxr-xr-x    3 dongdong  admin       96 Mar  1 14:00 python2.7
drwxr-xr-x    3 dongdong  admin       96 Mar  1 14:00 python3.7
lrwxr-xr-x    1 dongdong  admin       30 Mar  5 08:09 qt4 -> ../Cellar/qt@4/4.8.7_5/lib/qt4
lrwxr-xr-x    1 dongdong  admin       35 Mar  1 13:59 urdfdom -> ../Cellar/urdfdom/1.0.3/lib/urdfdom
lrwxr-xr-x    1 dongdong  admin       51 Mar  1 13:59 urdfdom_headers -> ../Cellar/urdfdom_headers/1.0.3/lib/urdfdom_headers

jcarpent, 2019-03-06:
You should do `brew switch collada-dom 2.4.4`

ddliugit, 2019-03-06:
I don't have that version

$ brew switch collada-dom 2.4.4
Error: collada-dom does not have a version "2.4.4" in the Cellar.
collada-dom installed versions: 2.4.0

ddliugit, 2019-03-06:
how do I install that version

jcarpent, 2019-03-06:
As usual, or by googling:
`brew upgrade collada-dom`

ddliugit, 2019-03-06:
Actually I tried this command, but it seems this is the latest version

$ brew upgrade collada-dom
Error: collada-dom 2.4.0 already installed

jcarpent, 2019-03-06:
brew install collada-dom --HEAD

ddliugit, 2019-03-06:
I think this command is used to install the latest version 2.5 instead of 2.4.4 

see the following error message:

 gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
qt.qpa.window: Window position QRect(208,-55 1023x1009) outside any known screen, using primary screen
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
   hpp-gui
   python-pinocchio
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/base.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/base.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/base.dae Desktop/models/ur_description/meshes/ur5/visual/base.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/base.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/forearm.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/forearm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/forearm.dae Desktop/models/ur_description/meshes/ur5/visual/forearm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/forearm.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae: file not handled
[... cut ...]
Node "world/pinocchio/visuals/shoulder_link_0" does not exist.
Node "world/pinocchio/visuals/upper_arm_link_0" does not exist.
Node "world/pinocchio/visuals/forearm_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_1_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_2_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_3_link_0" does not exist.
Node "world/pinocchio/visuals/tool0_0" does not exist.
Node "world/pinocchio/visuals/tool0_1" does not exist.
Node "world/pinocchio/visuals/tool0_2" does not exist.
()

ddliugit, 2019-03-06:
Also, I tried to check the directory for the newly installed Collada-dom:

It is like this

$ ls /usr/local/opt/collada-dom/lib/
cmake					libcollada-dom2.5-dp.dylib
libcollada-dom2.5-dp.0.dylib		pkgconfig
libcollada-dom2.5-dp.2.5.0.dylib

ddliugit, 2019-03-06:
The problem is how to install the certain version 2.4.4 you mentioned

jcarpent, 2019-03-06:
The problem is comming from homebrew itself. I will try to make a PR there.

jcarpent, 2019-03-06:
@ddliugit See Homebrew/homebrew-core#37658

ddliugit, 2019-03-06:
Do I need to reinstall it now

jcarpent, 2019-03-07:
@ddliugit You can now reinstall osg:
`brew reinstall open-scene-graph-with-colladadom`

ddliugit, 2019-03-07:
Do I need to update the Collada-dom version?

I tried, but it doesn't work 

Same problem happened again

$ gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
qt.qpa.window: Window position QRect(208,-55 1023x1009) outside any known screen, using primary screen
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
   hpp-gui
   python-pinocchio
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/base.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/base.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/base.dae Desktop/models/ur_description/meshes/ur5/visual/base.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/base.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/forearm.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/forearm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/forearm.dae Desktop/models/ur_description/meshes/ur5/visual/forearm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/forearm.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae: file not handled
[... cut ...]
Node "world/pinocchio/visuals/wrist_1_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_2_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_3_link_0" does not exist.
Node "world/pinocchio/visuals/tool0_0" does not exist.
Node "world/pinocchio/visuals/tool0_1" does not exist.
Node "world/pinocchio/visuals/tool0_2" does not exist.
^C
Dongdongs-MacBook-Pro:~ dongdong$ brew switch collada-dom 2.4.4
Error: collada-dom does not have a version "2.4.4" in the Cellar.
collada-dom installed versions: HEAD-d37ae75

ddliugit, 2019-03-07:
$ brew reinstall open-scene-graph-with-colladadom
==> Reinstalling gepetto/gepetto/open-scene-graph-with-colladadom 
==> Downloading https://github.com/Gepetto/OpenSceneGraph/releases/download/Open
Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/70cd19f617f7360a71789edb206def44abcd6d5d94d7d51460c587c5c71f7c4e--open-scene-graph-with-colladadom-3.6.3.mojave.bottle.tar.gz
==> Pouring open-scene-graph-with-colladadom-3.6.3.mojave.bottle.tar.gz
^CError: Directory not empty @ dir_s_rmdir - /var/folders/p6/h41q0rmj42lfjkf1zg7zb2rw0000gn/T/d20190307-42510-1rv0zsr
Warning: Bottle installation failed: building from source.
Error: An exception occurred within a child process:
  RuntimeError: /usr/local/opt/graphviz not present or broken
Please reinstall graphviz. Sorry :(
Dongdongs-MacBook-Pro:~ dongdong$ rm -r /Users/dongdong/Library/Caches/Homebrew/downloads/70cd19f617f7360a71789edb206def44abcd6d5d94d7d51460c587c5c71f7c4e--open-scene-graph-with-colladadom-3.6.3.mojave.bottle.tar.gz
Dongdongs-MacBook-Pro:~ dongdong$ brew reinstall open-scene-graph-with-colladadom
==> Reinstalling gepetto/gepetto/open-scene-graph-with-colladadom 
==> Downloading https://github.com/Gepetto/OpenSceneGraph/releases/download/Open
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws
######################################################################## 100.0%
==> Pouring open-scene-graph-with-colladadom-3.6.3.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3: 8,895 files, 229.7MB
Dongdongs-MacBook-Pro:~ dongdong$ ipython 
Python 2.7.15 |Anaconda, Inc.| (default, Dec 14 2018, 13:10:39) 
Type "copyright", "credits" or "license" for more information.

IPython 5.8.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: from pinocchio.robot_wrapper import RobotWrapper
   ...: 
   ...: pkg  = 'Desktop/models'
   ...: urdf = pkg + '/ur_description/urdf/ur5_gripper.urdf'
   ...: robot = RobotWrapper.BuildFromURDF(urdf,[pkg,])
   ...: 

In [2]: robot.initDisplay(loadModel=True)

In [3]:

ddliugit, 2019-03-07:
I think the problem is caused by the Collada dom version

jcarpent, 2019-03-07:
Sorry, I put the file at a wrong place. Can you retry:
`brew reinstall open-scene-graph-with-colladadom`

ddliugit, 2019-03-07:
Same issue:

$ gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
qt.qpa.window: Window position QRect(208,-55 1023x1009) outside any known screen, using primary screen
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
   hpp-gui
   python-pinocchio
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
BOX does not exist.: unspecified iostream_category error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/base.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/base.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/base.dae Desktop/models/ur_description/meshes/ur5/visual/base.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/base.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/shoulder.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/upperarm.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/forearm.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/forearm.dae could not be loaded
You may try to convert the file with the following command:
osgconv Desktop/models/ur_description/meshes/ur5/visual/forearm.dae Desktop/models/ur_description/meshes/ur5/visual/forearm.dae.osgb
File Desktop/models/ur_description/meshes/ur5/visual/forearm.dae found but could not be opened. Check that a plugin exist.
Warning: dynamic library '/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so' exists, but an error occurred while trying to open it:
dlopen(/usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so, 9): Library not loaded: /usr/local/opt/collada-dom/lib/libcollada-dom2.4-dp.0.dylib
  Referenced from: /usr/local/Cellar/open-scene-graph-with-colladadom/3.6.3/lib/osgPlugins-3.6.3/osgdb_dae.so
  Reason: image not found
Error reading file Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae: file not handled
File: Desktop/models/ur_description/meshes/ur5/visual/wrist1.dae could not be loaded
[... cut ...]
Node "world/pinocchio/visuals/shoulder_link_0" does not exist.
Node "world/pinocchio/visuals/upper_arm_link_0" does not exist.
Node "world/pinocchio/visuals/forearm_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_1_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_2_link_0" does not exist.
Node "world/pinocchio/visuals/wrist_3_link_0" does not exist.
Node "world/pinocchio/visuals/tool0_0" does not exist.
Node "world/pinocchio/visuals/tool0_1" does not exist.
Node "world/pinocchio/visuals/tool0_2" does not exist.

jcarpent, 2019-03-07:
Can you do:
`brew uninstall open-scene-graph-with-colladadom`first, and then 
`brew install open-scene-graph-with-colladadom`
And also provides the output of
`ls /usr/local/opt/collada-dom/lib`

ddliugit, 2019-03-07:
The problem still exists due to the lack of colladamdom version 2.4 referred by osg

jcarpent, 2019-03-07:
Yes, but the new bottle version of `open-scene-graph-with-colladado` is linking to collada-dom 2.5

ddliugit, 2019-03-07:
Yes, it worked now.

Thanks a lot!

ddliugit, 2019-03-07:
But I found this error after I tried to move the object in the window

visualObj  = robot.visual_model.geometryObjects[4] # 3D object represent
   ...: ing the robot forarm
   ...: visualName = visualObj.name                        # Name associated to 
   ...: this object
   ...: visualRef  = robot.viewerNodeNames(visualObj)      # Viewer reference (s
   ...: tring) representing this object
   ...: 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-89a97a7b3eca> in <module>()
      1 visualObj  = robot.visual_model.geometryObjects[4] # 3D object representing the robot forarm
      2 visualName = visualObj.name                        # Name associated to this object
----> 3 visualRef  = robot.viewerNodeNames(visualObj)      # Viewer reference (string) representing this object

AttributeError: 'RobotWrapper' object has no attribute 'viewerNodeNames'

jcarpent, 2019-03-07:
@ddliugit Please send a clear and self-contained example, otherwise it will be a mess to debug this kind of stuff.

ddliugit, 2019-03-07:
http://homepages.laas.fr/nmansard/teach/robotics2018/tp1.html

jcarpent, 2019-03-07:
Can you simply provide a self content *.py file that highlight this behavior? Thanks.

ddliugit, 2019-03-07:
[example.txt](https://github.com/stack-of-tasks/pinocchio/files/2941772/example.txt)

[5 more replies]