stack-of-tasks/pinocchio #715: centerOfMass<bool do_position, bool do_velocity, bool do_acceleration>

gabrielebndn, 2019-03-05:
Pinocchio 1 had a template function called `centerOfMass<bool do_position, bool do_velocity, bool do_acceleration>(model,data)`
It has been replaced by `centerOfMass(model,data,LEVEL)`, plus a `centerOfMass(model,data)` which basically defaults to `LEVEL=2`.
May I know why? I think the old interface was more flexible and should be restored.

jcarpent, 2019-03-05:
The two options provide the same results, with similar performances. The second signature was simpler to handle and to bind.

gabrielebndn, 2019-03-05:
Actual use case: an algorithm fills com and vcom as side-effect. But we also need acom. The old signature provided a way to compute acom without re-computing com and vcom. The new one does not.
But I do agree the Python signature would come out as a mess.

jcarpent, 2019-03-05:
But you need the vcom to compute the acom, and so on

jcarpent, 2019-03-05:
One thing which can be done is to add an enum which will define `POSITION=1`, `VELOCITY=2`, 'ACCELERATION=4` and `POSITION | ACCELERATION` will just proceed to the position and acceleration updates.

gabrielebndn, 2019-03-08:
Finally, I think it's not really worth it. The current interface is intuitive enough, changing it would just complicate it and the gains would be negligible. I'll close this issue.

closed 2019-03-08 by gabrielebndn