stack-of-tasks/pinocchio #717: mprint issue on os

ddliugit, 2019-03-05:
When I run mprint, I found the following problem, can you help me solve it?

In [17]: mprint(rand([6,6]))       # Matlab-style print
ans  = 

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-17-35cbe44e74d9> in <module>()
----> 1 mprint(rand([6,6]))       # Matlab-style print

/usr/local/lib/python2.7/site-packages/pinocchio/utils.pyc in mprint(M, name, eps)
     78     Mmin = lambda M: M.min() if np.nonzero(M)[1].shape[1]>0 else M.sum()
     79     Mmax = lambda M: M.max() if np.nonzero(M)[1].shape[1]>0 else M.sum()
---> 80     Mm = Mmin(abs(M[np.nonzero(M)]))
     81     MM = Mmax(abs(M[np.nonzero(M)]))
     82 

/usr/local/lib/python2.7/site-packages/pinocchio/utils.pyc in <lambda>(M)
     76     print()
     77 
---> 78     Mmin = lambda M: M.min() if np.nonzero(M)[1].shape[1]>0 else M.sum()
     79     Mmax = lambda M: M.max() if np.nonzero(M)[1].shape[1]>0 else M.sum()
     80     Mm = Mmin(abs(M[np.nonzero(M)]))

IndexError: tuple index out of range

jcarpent, 2019-03-05:
I solved this issue with 33ed834f13d42ff66661111cf2c3601227236a42
Solving this issue was really easy. I think you should think to solve some of them by yourself: it allows you to get into Pinocchio very efficiently, and it will the supporting team too.

jcarpent, 2019-03-05:
you can reinstall Pinocchio from the devel branch using:
```brew reinstall pinocchio --devel```

closed 2019-03-05 by jcarpent

ddliugit, 2019-03-05:
Sorry, it seems not working for me

$ brew reinstall pinocchio --devel
Usage: brew reinstall [options] formula

Uninstall and then install formula (with existing and any appended install options).

Unless HOMEBREW_NO_INSTALL_CLEANUP is set, brew cleanup will be run for the reinstalled formulae or, every 30 days, for all formulae.

    -d, --debug                      If brewing fails, open an interactive
                                     debugging session with access to IRB or a
                                     shell inside the temporary build directory
    -s, --build-from-source          Compile formula from source even if a
                                     bottle is available.
        --force-bottle               Install from a bottle if it exists for the
                                     current or newest version of macOS, even if
                                     it would not normally be used for
                                     installation.
        --keep-tmp                   Don't delete the temporary files created
                                     during installation.
    -f, --force                      Install without checking for previously
                                     installed keg-only or non-migrated
                                     versions.
    -v, --verbose                    Print the verification and postinstall
                                     steps.
        --display-times              Print install times for each formula at the
                                     end of the run.
        --without-python             pinocchio: Build without Python support
        --without-fcl                pinocchio: Build without FCL support
        --without-boost-python       pinocchio: Build without boost-python
                                     support
        --without-urdfdom            pinocchio: Build without urdfdom support
        --without-eigenpy            pinocchio: Build without eigenpy support
        --without-python@2           pinocchio: Build without python@2 support
        --without-numpy              pinocchio: Build without numpy support
        --without-hpp-fcl            pinocchio: Build without hpp-fcl support
    -h, --help                       Show this message.
Error: invalid option: --devel

ddliugit, 2019-03-05:
I don't know how to revise the source code and install it to my computer. 

I am very interested into contributing to this software, especially for the installation tutorial on Mac

Can you tell me the procedure of revising code in GitHub

jcarpent, 2019-03-05:
You should then do 
`brew uninstall pinocchio` followed by
`brew install pinocchio --devel`

ddliugit, 2019-03-05:
I met this problem after doing this command:

$ brew uninstall pinocchio
Uninstalling /usr/local/Cellar/pinocchio/2.1.0... (1,845 files, 31.2MB)

$ brew install pinocchio --devel
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
dockerize                                zydis
==> Updated Formulae
akamai              dav1d               jo                  thrift
apktool             dmenu               libpq               unzip
at-spi2-atk         dwm                 opentracing-cpp     wartremover
at-spi2-core        erlang              osquery             weboob
bazel               jadx                pulumi
cash-cli            jenkins             scw

Error: No devel block is defined for gepetto/gepetto/pinocchio

jcarpent, 2019-03-05:
I've just pushed. Could you try it again?

ddliugit, 2019-03-05:
Problem changed:

$ brew install pinocchio --devel
Error: invalid attribute for formula 'gepetto/gepetto/pinocchio': version (nil)
Please report this bug:
  https://docs.brew.sh/Troubleshooting
/usr/local/Homebrew/Library/Homebrew/formula.rb:257:in `validate_attributes!'
/usr/local/Homebrew/Library/Homebrew/formula.rb:204:in `initialize'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:112:in `new'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:112:in `get_formula'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:306:in `factory'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:50:in `block in formulae'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:48:in `map'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:48:in `formulae'
/usr/local/Homebrew/Library/Homebrew/cli_parser.rb:155:in `formula_options'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:86:in `block in install_args'
/usr/local/Homebrew/Library/Homebrew/cli_parser.rb:34:in `instance_eval'
/usr/local/Homebrew/Library/Homebrew/cli_parser.rb:34:in `initialize'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:15:in `new'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:15:in `install_args'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:91:in `install'
/usr/local/Homebrew/Library/Homebrew/brew.rb:102:in `<main>'

jcarpent, 2019-03-05:
Brew has changed. So the new policy is now:
```brew install pinocchio --HEAD```

ddliugit, 2019-03-05:
Built failed:

dongdong$ brew install pinocchio --HEAD
==> Auto-updated Homebrew!
Updated 1 tap (gepetto/gepetto).
==> Updated Formulae
gepetto/gepetto/gepetto-viewer ✔
Updating Homebrew...

==> Installing pinocchio from gepetto/gepetto
==> Cloning https://github.com/stack-of-tasks/pinocchio.git
Cloning into '/Users/dongdong/Library/Caches/Homebrew/pinocchio--git'...
==> Checking out branch devel
Already on 'devel'
Your branch is up to date with 'origin/devel'.
Submodule '.travis' (https://github.com/jrl-umi3218/jrl-travis) registered for path '.travis'
Submodule 'cmake' (https://github.com/jrl-umi3218/jrl-cmakemodules) registered for path 'cmake'
Submodule 'models/romeo' (https://github.com/laas/romeo.git) registered for path 'models/romeo'
Cloning into '/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/.travis'...
Cloning into '/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/cmake'...
Cloning into '/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/models/romeo'...
Submodule path '.travis': checked out '083fa2cb0fa4ad594926d9bb3d246075e62ce9ee'
Submodule path 'cmake': checked out '1d9aeca25e970d2d967fd5be0fb93fe961db121b'
Submodule path 'models/romeo': checked out '200b93eb20d0c3c95d55c29fb47bb87eb343f586'
Submodule 'romeo_description/cmake' (git://github.com/jrl-umi3218/jrl-cmakemodules.git) registered for path 'models/romeo/romeo_description/cmake'
Cloning into '/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/models/romeo/romeo_description/cmake'...
Submodule path 'models/romeo/romeo_description/cmake': checked out 'a831f5403a93b67c6cb8675625e79dd676a565f3'
/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/.travis
/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/cmake
/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/models/romeo
/Users/dongdong/Library/Caches/Homebrew/pinocchio--git/models/romeo/romeo_description/cmake
==> git submodule update --init
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG
==> make
Last 15 lines from /Users/dongdong/Library/Logs/Homebrew/pinocchio/03.make:
make[1]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 48%] Linking CXX executable sample-models
cd /tmp/pinocchio-20190305-10432-1q48mi3/build/unittest && /usr/local/Cellar/cmake/3.13.4/bin/cmake -E cmake_link_script CMakeFiles/sample-models.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  -L/usr/local/Cellar/assimp/4.1.0/lib -L/usr/local/Cellar/octomap/1.9.0/lib -L/usr/local/Cellar/hpp-fcl/1.0.1/lib -lhpp-fcl /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib -loctomap -loctomath -lassimp CMakeFiles/sample-models.dir/sample-models.cpp.o  -o sample-models  -L/usr/local/Cellar/assimp/4.1.0/lib  -L/usr/local/Cellar/octomap/1.9.0/lib  -L/usr/local/Cellar/hpp-fcl/1.0.1/lib -Wl,-rpath,/usr/local/Cellar/assimp/4.1.0/lib -Wl,-rpath,/usr/local/Cellar/octomap/1.9.0/lib -Wl,-rpath,/usr/local/Cellar/hpp-fcl/1.0.1/lib -Wl,-rpath,/tmp/pinocchio-20190305-10432-1q48mi3/build/src -L/usr/local/Cellar/assimp/4.1.0/lib -L/usr/local/Cellar/octomap/1.9.0/lib -L/usr/local/Cellar/hpp-fcl/1.0.1/lib -lhpp-fcl /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib -loctomap -loctomath -lassimp /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib ../src/libpinocchio.dylib /usr/local/lib/libboost_unit_test_framework-mt.dylib -L/usr/local/Cellar/assimp/4.1.0/lib -L/usr/local/Cellar/octomap/1.9.0/lib -L/usr/local/Cellar/hpp-fcl/1.0.1/lib -L/usr/local/Cellar/console_bridge/0.4.3/lib -L/usr/local/Cellar/urdfdom/1.0.3/lib -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge -lhpp-fcl /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib -loctomap -loctomath -lassimp /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib -loctomap -loctomath -lassimp 
[ 48%] Built target sample-models
[ 48%] Linking CXX executable centroidal
cd /tmp/pinocchio-20190305-10432-1q48mi3/build/unittest && /usr/local/Cellar/cmake/3.13.4/bin/cmake -E cmake_link_script CMakeFiles/centroidal.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  -L/usr/local/Cellar/assimp/4.1.0/lib -L/usr/local/Cellar/octomap/1.9.0/lib -L/usr/local/Cellar/hpp-fcl/1.0.1/lib -lhpp-fcl /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboos [... cut ...] -mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib -loctomap -loctomath -lassimp /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib -loctomap -loctomath -lassimp 
[ 48%] Built target geom
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/gepetto/homebrew-gepetto/issues

jcarpent, 2019-03-05:
@ddliugit I have updated the Formula. You can retry again. It should work fine now.

ddliugit, 2019-03-05:
It worked!!

Thanks