stack-of-tasks/pinocchio #720: Error about omniORB.CORBA.BAD_PARAM?

littleggghost, 2019-03-06:
Hi, there
I test the tutorial in [4.1) Direct dynamics](http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-practical-exercises_4-dyn.html)
and got an error:
```
Traceback (most recent call last):
  File "exercise_4_1_move.py", line 15, in <module>
    robot.display(robot.q0)
  File "/home/zyg/pinocchio/doc/d-practical-exercises/src/robot_hand.py", line 316, in display
    visual.place(self.viewer, self.data.oMi[visual.jointParent])
  File "/home/zyg/pinocchio/doc/d-practical-exercises/src/robot_hand.py", line 36, in place
    display.place(self.name, oMbody, False)
  File "/home/zyg/pinocchio/doc/d-practical-exercises/src/display.py", line 53, in place
    se3ToXYZQUAT(M))
  File "/opt/openrobots/lib/python2.7/site-packages/gepetto/corbaserver/graphical_interface_idl.py", line 393, in applyConfiguration
    return _omnipy.invoke(self, "applyConfiguration", _0_gepetto.corbaserver.GraphicalInterface._d_applyConfiguration, args)
omniORB.CORBA.BAD_PARAM: CORBA.BAD_PARAM(omniORB.BAD_PARAM_WrongPythonType, CORBA.COMPLETED_NO)
```
This error not happens with v2.0.0, and occurs with v2.1.0.

jmirabel, 2019-03-06:
Related to https://github.com/stack-of-tasks/pinocchio/pull/702.

Try to replace `se3ToXYZQUAT(M)` by `se3.se3ToXYZQUATtuple(M)`.

littleggghost, 2019-03-06:
Thanks. I think some docs should be updated.

closed 2019-03-06 by littleggghost

jcarpent, 2019-03-06:
@jmirabel Thanks for this quick reply.
@littleggghost Thanks for raising this issue.
I'm fixing it on the devel branch.