stack-of-tasks/pinocchio #726: Visualization of sample model not working

gabrielebndn, 2019-03-07:
The following minimal example
```
import pinocchio as pin

model = pin.buildSampleModelHumanoid()
geom_model = pin.buildSampleGeometryModelHumanoid(model)
q = pin.neutral(model)

robot = pin.RobotWrapper(model,geom_model,geom_model)

raw_input("initDisplay")
robot.initDisplay()

raw_input("loadDisplayModel")
robot.loadDisplayModel()

raw_input("display")
robot.display(q)

print("done")
```
is not working. However, gepetto-gui itself seems to be fine (`examples/python/gepetto-viewer.py`, which loads DAE files for romeo, is working for me).

On `loadDisplayModel()`, I get a bunch of errors like this
```
 does not exist.: iostream error
```
followed by errors that look like
```
Node "world/pinocchio/collisions/rlegshoulder_object" does not exist.
```

I don't know where this problem might come from. Am I doing this right? is there something wrong in `buildSampleGeometryModelHumanoid`? Can this be a probem of hpp-fcl instead of Pinocchio?

jcarpent, 2019-03-07:
I don't think that hpp-fcl basic geometries are parsed by the viewer.

gabrielebndn, 2019-03-07:
I don't think there is any parsing involved. Plus, I remember @nmansard made the `buildSampleGeometryModel` methods, and they were working on gepetto-viewer. If it does not, I think it should. @nmansard, may you verify this?

nmansard, 2019-03-07:
robot wrapper does not create the simple geometry. And it cannot do it because we don't have the bindings for the simple geometries.

I started to map the simple geometries to the GUI, but manually. And I stopped in the middle.

@jcarpent I think you told me that you had some piece of code to expose hpp-fcl simple geometries. I believe we even have an issue for that open since a long time.

gabrielebndn, 2019-03-08:
The issue was #548, but it was closed, I don't know why.

jcarpent, 2019-03-08:
I'm not able to find them. They seem to be on my previous laptop, which has been reformated since then :(

gabrielebndn, 2020-01-08:
Solved by #1004

closed 2020-01-08 by gabrielebndn