stack-of-tasks/pinocchio #729: Optinal dependency on hpp-fcl but fails to build if not found

jviereck, 2019-03-07:
Trying to compile the code (master=db8fe0add7d10ea31) from source on 16.04 Ubuntu for python2, I get the error:

```
In file included from /home/jviereck/dev/tools/pinocchio/build27/include/pinocchio/parsers/urdf.hpp:268:0,
                 from /home/jviereck/dev/tools/pinocchio/build27/include/pinocchio/bindings/python/parsers/parsers.hpp:14,
                 from /home/jviereck/dev/tools/pinocchio/bindings/python/parsers/expose-parsers.cpp:5:
/home/jviereck/dev/tools/pinocchio/build27/include/pinocchio/parsers/urdf/geometry.hxx:294:51: error: ‘pinocchio::fcl::MeshLoaderPtr’ has not been declared
                                              fcl::MeshLoaderPtr& meshLoader,
                                                   ^
/home/jviereck/dev/tools/pinocchio/build27/include/pinocchio/parsers/urdf/geometry.hx
```

Note that it looks like my `cmake ..` configuration doesn't find the optinal hpp-fcc package. If the package is optional, I wonder why the compilation is breaking still.

jcarpent, 2019-03-08:
@jviereck Thanks for reporting this bug. I've proposed a fix for it #730.

jcarpent, 2019-03-08:
Fixed by #730.

closed 2019-03-08 by jcarpent