stack-of-tasks/pinocchio #731: robot.collisionPairs.append() can't attach witness points to body?

littleggghost, 2019-03-08:
Hi
In chapter [4.3) Collision checking](http://projects.laas.fr/gepetto/doc/stack-of-tasks/pinocchio/master/md_doc_d-practical-exercises_4-dyn.html)
After I add the witness points, I find they did't move with links, here the pics:
before apply torque:
[image]
after apply torque:
[image]

jcarpent, 2019-03-08:
@nmansard As you are the author of these exercises, can you look at this display problem?

littleggghost, 2019-04-16:
Hi, @jmirabel 
Have you ever meet this problem? Hope you can give me an advice, thanks.

littleggghost, 2019-04-16:
Finally, I found function robot.visuals.append() works.

closed 2019-04-16 by littleggghost