stack-of-tasks/pinocchio #741: Inertial information in frames

gabrielebndn, 2019-03-21:
In URDF, links are systematically connected by joints. Inertial information is attached to links. Each joint has exactly one parent link and one child link. Joints may be fixed.

In Pinocchio, however, there are no fixed joints, and inertial information is associated directly to the (movable) joints.

When parsing an URDF joint, there are basically two main cases:
- The joint is an *actual*, configurable joint. In this case, the joint is simply added to the kinematic chain, and the inertia of its child link is added to the vector of inertias. A `Frame` is also added for the joint (of type `JOINT`), and one for the link (of type `BODY`).
- The joint is a fixed joint. In this case, a joint is not added to the kinematic chain. Instead, only the two `Frame` objects are added (of type `FIXED_JOINT` for the joint), in order to preserve information about their placement. In order to preserve information about the connectivity of the original tree structure, `Frame` objects also contain a `previousFrame` element. The inertia of the link is simply added to that of the parent joint.

In the second case, during the process, inertial information associated to the individual link is lost. Only the aggregated inertia is retained. However, sometimes, this information is needed, to know e.g. how much weight is carried by a force sensor.

Would it be possible to retain the information? For instance, by adding inertia inside `Frame` objects, at least for those of the type `BODY`?

jcarpent, 2019-03-21:
I think you should first think why you really need to keep the inertia of a fixed link?
Is it because you want to know exactly the inertia of the torque sensor unit?

In this context, I think you should have a dedicated way to retrieve this information from URDF.

A frame only contains geometric information on purpose, it is not a quantity which contains a dynamical quantity. For instance, for geometries, we may think to keep all the visual information that we are not providing.

One solution is to derive a class BodyFrame from FrameTpl which may contain the related information. Then, it is sufficient to do a dynamic cast to retrieve the type you want to deal with in your precise context. Is this solution clear enough?

gabrielebndn, 2019-03-21:
@jcarpent I was thinking something like your last suggestion. However, I realize it might not only add complication to the code, but also add a lot of redundant information. It might end up just bloating the `Model` class.
I am open to the idea of a dedicated way to retrieve the information from the URDF.

jcarpent, 2019-03-27:
@gabrielebndn Do you want to implement the proposed solution?

gabrielebndn, 2019-03-27:
@jcarpent which solution are you talking about?

> a dedicated way to retrieve this information from URDF

or

> to derive a class BodyFrame from FrameTpl which may contain the related information

The more I think about it, the more I am in favour of the first solution. Adding inertia information to each `BODY_FRAME` would add a lot of redundant information and make a model heavier, for something which is not super-essential. A way to reduce this problem would be to only add this information for frames of type `FIXED_JOINT`.

But in both cases there are technical complications. It is not possible to put both `Frame` objects and its subclasses inside the `std::vector` `model.frames`, for instance. One should either use pointers (which is not the current paradigm of Pinocchio and I would definitely avoid) or implement a variant mechanism such as the one existing for the joint types, which would complicate the design, probably further increase compilation time, ad possibly impact compilation time a bit too.

Therefore, I think the best is to add a method called `getInertiaFromUrdf(model, filename, bodyname)` to [src/parsers/urdf/model.hxx](https://github.com/stack-of-tasks/pinocchio/blob/devel/src/parsers/urdf/model.hxx) which simply returns the inertia (the model parameter would only be there for type inference of the return value and for e.g. checking that a body with that name exists).

closed 2019-04-16 by jcarpent

duburcqa, 2021-02-13:
I'm trying to recompute some data that are provided by Mujoco simulator using Pinocchio. In particular, it requires computing the vector of subtree center of mass positions for each bodies, separately for bodies connected by fixed joints. Currently, it is not possible compute such quantity since in this case the inertias are merged in a single one. What do you think would be the best approach to deal with this issue ?

jcarpent, 2021-02-14:
Could you elaborate rate more on why you need this sub tree com? Which quantity requires it?

duburcqa, 2021-02-15:
Yes of course, are you familiar with Mujoco ? I'm recomputing [data.cinert](http://www.mujoco.org/book/APIreference.html). It is defined as the "com-based body inertia and mass", which is in fact some weird stuff: It corresponds to the inertia and mass of the body in world frame, and the position of body's subtree com relative to robot's com (local world aligned).

So far so good and I able to recompute this quantity without issue for bodies not connected via fixed joints. But obviously it does not work in the later case, since the body's subtree com is not defined in such a case.

jcarpent, 2021-04-01:
Solved by #1415.