stack-of-tasks/pinocchio #742: Fusion of Models

jmirabel, 2019-03-21:
I would like to be able to append a Model to a specific joint of another Model. I cannot think of any easy way of doing it.

Using Pinocchio, I would have to write:
- an algo like buildModel. That's the easy part.
- an algo that reorder the joints so that all the children of a joint are contiguous. This looks a bit more tricky as they are many fields to be updated, both in Model and GeomModel. it's going to be a nightmare to maintain...

I could also create a new Model and append joints to it in the right order. It seems a bit easier.

Any hints or comments on the above ?

jcarpent, 2019-03-21:
>     * an algo that reorder the joints so that all the children of a joint are contiguous. This looks a bit more tricky as they are many fields to be updated, both in Model and GeomModel. it's going to be a nightmare to maintain...

You should follow this strategy if you want Pinocchio to be efficient and it is mandatory in the context of CRBA algorithms for instance.
I would say that is better to create a new model from two existing ones. It is just a call to appendJoint method, which is pretty sufficient.

jmirabel, 2019-03-21:
For my application, it doesn't have to be efficient.

I will have a look at the second option.

jcarpent, 2019-03-21:
I would prefer to implement directly the final solution, which means having continuous ordering for directly parent joints. It will avoid additional work later and full compatibility with all the existing algo.

jmirabel, 2019-03-22:
> I would prefer to implement directly the final solution

what do you mean ?

jcarpent, 2019-03-22:
I mean, the solution of merging two models in a new model, having all the index of a given subtree represented by a continuous collection (again, this is mandatory for most dynamics algo).

jmirabel, 2019-03-27:
In which file should this be ? My first thought is `algorithm/model.hpp`.

jmirabel, 2019-03-27:
It could also go into `algorithm/copy.hpp`.

jmirabel, 2019-03-27:
It could also be a constructor of Model class. It may be a little awkward though.

gabrielebndn, 2019-03-27:
Related to #470.
A new file called `algorithm/model.hpp` could contain all algorithms related to directly manipulating the model, such as the proposed fusion of models, the extraction of subtrees, and possibly joint reordering (see #379).
My feeling though is that all these features would be best implemented as instance methods of the `Model` class.

jcarpent, 2019-03-27:
I would do it in `multibody/model.hxx`

jcarpent, 2019-03-28:
solved by #752.

closed 2019-03-28 by jcarpent